pascallanger/DIY-Multiprotocol-TX-Module #2: Move Binaries to GH-"Releases" Tab

evils, 2016-01-01:
Afaik the Binaries folder in the repos root holds compiled binaries from the source.

For best practice as well as to minimize the size of this repo, avoid confusion and to keep things together. 
I would suggest to remove that folder (maybe even extinguish the folder from the GH history to avoid pulling that info when cloning the repo) and add those binaries to a Release.
A GitHub release would allow to mark the commit from which these binaries have been created from and help to identify the corresponding source.

Benefits over the Binaries folder are also:
- Release Info eliminates additional Readme in Binaries folder
- Connection between corresponding code and binary
- Repo size would be smaller
- Clarity for Users trying to get binaries

pascallanger, 2016-01-01:
Hi Elvis,
I've relocated the hex file under Release. Hopefully I've done it correctly.
I've also tried to "extinguish the folder from the GH history".
Let me know if everything looks ok.
Thanks, Pascal

closed 2016-01-01 by pascallanger

evils, 2016-01-02:
@pascallanger 
Looks good to me! You might want to name the actual release different from the product, but that's only a choice ;)

The hex files are still in the git history (see de4c841961fcca951afb2407b43b8c2cfa19d141), you would need to rewrite your history and fast forward/force the rewritten history to completely extinguish these hex files. (I normally use smth like BFG repo cleaner for such jobs) 
After all, they are not huge so that it would be a significant improvement.