pascallanger/DIY-Multiprotocol-TX-Module #9: Bind SFHSS failed！

zyren, 2016-04-27:
Hi Pascal,

I use the last code and I don't bind, with anithing futaba receivers, and WL_TOYS XK110 failed too.
Expect improvement,thanks.

regards.

zyren, 2016-04-27:
_config.h:

/*1*/	{MODE_SFHSS,	0,	0,	P_HIGH	,	NO_AUTOBIND,	0},

pascallanger, 2016-05-03:
Hi Zyren,
I've already tried to check a lot of things but I can't see what's going wrong.
I don't have any SFHSS RX to test myself... So unless someone sends me something to work on, I don't see how to solve this issue.
- Pascal

anshengchang, 2016-05-26:
I have the same issue.
Who could help me!!!
pascallanger, Where do you live???

pascallanger, 2016-05-26:
I leave in France.
- Pascal

zyren, 2016-05-31:
Hi Pascallanger,
I saw someone say deviationTx, the options set between +25 to +95, perfect binding XK110 and XK120 ,, I will try tomorrow, then come back to report the results.
regards,
zyren.

qdog2012, 2016-08-12:
hi zyren, how is your test result? i have this issue too.

qdog2012, 2016-08-12:
Multiprotocol code version：2.1.8 mult 0803
TX type: x9d plus
Using PPM or Serial: serial
Explanation of the behavior and reproduction steps: can not bind xk120 with sfhss protocol. using 4in1 module.

qdog2012, 2016-08-12:
Multiprotocol code is latest version

pascallanger, 2016-08-14:
Have you tried to shift the frequency using the option value? You need to try all the values...But anyway if there is an issue I don't have any RX to test the code on.
- Pascal

qdog2012, 2016-08-14:
i have tried to shift the frequency using the all option values, still no work.
i have a D10 tx, using deviation, it work for me.

pascallanger, 2016-08-14:
I'll check once again but I have little hope without a RX...

pascallanger, 2016-08-15:
I've just posted a new source version. Try to compile and test.

qdog2012, 2016-08-15:
thank you, but i can't see the new commit or new code in the github? 
Latest commit 9e902a5  8 days ago

pascallanger, 2016-08-15:
I'm on my mobile connection so syncing is slow, retry.

qdog2012, 2016-08-15:
okay, i got the new code. i'll try it later and post my test result. thanks again.

qdog2012, 2016-08-15:
hi pascallanger, i have tested the new code, still no work.
i realize when i using the ppm mode and s-fhss protocol, press and hold the bind button, the red light is turn off, unless release the bind button, is that mean the tx not going start to bind?

qdog2012, 2016-08-15:
is any thing i can do for help?

pascallanger, 2016-08-15:
Hold on. The TX does not enter bind mode, it's only the RX which binds with any TX around it.
Pressing the bind button will always turn off the LED since you shortcut it, this is not the program doing it but you are.
Have you really tried all the option values from -127 to +127 and try to bind the RX each time you are making a change???

qdog2012, 2016-08-15:
yes, i tried all the option values from -127 to +127 bind the RX each time making a change with SPI mode. my RX is XK120.

pascallanger, 2016-08-15:
On the Devo, have you changed fine or coarse to get it to work?

qdog2012, 2016-08-15:
on the Devo, my version is no fine or coarse option, only bind button in the GUI.

pascallanger, 2016-08-16:
I've just published a new version, please compile, try with different option values and report.
Thanks, Pascal

qdog2012, 2016-08-16:
ok, i'll report test result when i tried all option values.

pascallanger, 2016-08-16:
I've just published a new version (again), please compile, try with different option values and report.
Thanks, Pascal

qdog2012, 2016-08-16:
bad news, the latest code is no work.
i have order a new single cc2500 module, may be only the 4in1 module do not work.

pascallanger, 2016-08-16:
Really... Couple of questions:
 - Have you tested the latest version posted 2 hours ago?
 - Have you tested your CC2500 with another protocol to make sure it works? FrSky or FrSkyX?
 - Could you try to increase the delay line 104 : delayMicroseconds(6);  try to change the 6 by 100, 200,...

pascallanger, 2016-08-16:
FYI, people on the RCG forum have reported it to at least bind so it should do the same for you. But it's still unstable. I would like to test changing the delay...

qdog2012, 2016-08-17:
yeah, i can bind with RX, looks bind needs the RX so close. after RX bind ok. i have no any action, the red light is blink and irregular, since always getting TX signal. my option values is 18 to 95.
when i control the TX, the RX is slow reaction and not smooth.

pascallanger, 2016-08-17:
It should be solved on the latest version please confirm.

closed 2016-08-17 by pascallanger