pascallanger/DIY-Multiprotocol-TX-Module #12: Won't bind anymore

techcoast, 2016-07-28:
I had this little guy working with a CX10 and it worked great. I tried to copy the settings from the H8 example and aside from a couple of text errors, I think I got it all squared away, but the bind no longer works to the CX. I even went back to scratch and a basic setup but nothing happens.. any ideas?

pascallanger, 2016-07-28:
Hi Techcoast,
Not sure what you are talking about: "I tried to copy the settings from the H8 example and aside from a couple of text errors, I think I got it all squared away".
What's your setup?
Have you upgraded the firmware in  between?
You should be able to start from a clean model using AETR for the channel order and fly right away.
- Pascal

techcoast, 2016-07-28:
Hey Pascallenger, I put this in the wrong forum. :) Sorry about that! Part was a setting issue, and the other part was using a different stick position that was very different than the one that first worked. It's all been sorted. Thanks for the reply !

closed 2016-07-28 by pascallanger