pascallanger/DIY-Multiprotocol-TX-Module #13: Unable to bind to Inductrix (DSM2/X) on recent versions (ppm or serial)

Gaurhoth, 2016-08-17:
Multiprotocol Version: 1.1.2 (also tried 1.1.1)
TX: Taranis + Multiprotocol Board from Banggood (frsky version). It's the newer version with resistors on board.
Tried both PPM and Serial
Taranis OpenTX Version: 2.1.8 from http://plaisthos.de/opentx/ (07/23 and 08/16 versions)

When I originally received the board, I was able to successfully bind using PPM with the original Banggood firmware. Wanting to use Serial, I soldered the rx/tx pads as directed (resistors are already on the board) and upgraded to 1.1.2. Unsuccessful in binding despite trying NUMEROUS combinations of channel ranges/PPM timing. Inductrix never bound to the transmitter. Green and RED LED solid after bootup. Attempted binding with button and in the UI. No success.

Downgraded back to the orignal banggood firmware (fortunately took a backup) and unit now binds in PPM mode again.

Willing to work with someone to help troubleshoot.

pascallanger, 2016-08-17:
Hi Gaurhoth,
Couple of things:
 - Don't loose your time trying every possible options. The only thing that should matter in serial mode is the number of channels/frame rate.
 - In serial when you press the bind button on the GUI, you should see the module blinking for a few seconds and then return to solid, do you confirm?
 - Binding order should be:
    1. power tx and select the correct model with MULTI/DSM2 and the number of channels (6@11ms is usually a good spot)
    2. power inductrix in bind mode
    3. Press bind from the gui
    4. if it does not work, what's the LED status on the quad? any blinking change?
 - Can you build, test (changing number of channels/frame) with the latest github sources and report result (including LED pattern(s))?
 - Can you build, test (changing number of channels/frame) with the latest github sources with TELEMETRY disabled (commented) in _Config.h and report result (including LED pattern(s))?
Thanks, Pascal

Gaurhoth, 2016-08-22:
I resoldered the tx/rx pads and its working now. Apparently I had a bad solder connection.

closed 2016-08-22 by Gaurhoth