pascallanger/DIY-Multiprotocol-TX-Module #22: Unable to control WLToys A959 after binding.

mattfairfield856, 2016-11-01:
After binding to the RC car i cannot seem to get the controls working.  Are your protocols not meant to work with RC cars?

mattfairfield856, 2016-11-01:
We are also having the same problem with the sky matrix drone of binding but not able to control. (drone using YD717_NRF24L01_INO) (RC using V2X2_NRF24L01_INO)

pascallanger, 2016-11-02:
First, what type of remote control do you have, what module type (home made, 4-in-1,...), how is it connected Serial or PPM?
Do you have at least a model working?
From what you are saying I have the impression that you are using PPM mode and selecting the protocol using the dial. The fact that bind works indicate that the module works, the facts that you have no control indicates that the PPM signal is not getting to the module.
Pascal

mattfairfield856, 2016-11-02:
I am using the 4-in-1 radio chip, PPM mode, and I am trying to create the
PPM signal programmatically using an arduino uno instead of a controller. I
have found a PPM generation sketch example that im working on for the Uno
to send the PPM signal out of pin 10 to the PPMin pin on the 4-in-1. If I
increase channels 1 and 2 to about 1200 us the servos start running but
then reducing them does not shut them back down. Does every protocol work
with a serial connection too or just a certain few? (im using
V2X2_NRF24L01_INO and YD717_NRF24L01_INO specifically)

pascallanger, 2016-11-03:
Hi,
This is the wrong place to talk please ask your questions on the dedicated forum: https://www.rcgroups.com/forums/showthread.php?t=2165676&page=407 .
What I can tell you is that the problem is on your side with the PPM signal you are generating.
All protocols are working either using the serial or ppm mode.
Pascal

closed 2016-11-03 by pascallanger