pascallanger/DIY-Multiprotocol-TX-Module #25: Bug in the config with DSM and DEVO

tipouic, 2016-11-30:
If you comment the TELEMETRY, you have an variable not defined on the line 414 and 496.
If you comment the CYRF6936_INSTALLED int the config, you have an error with the line 162 and 294 in th e DEVO protocol. I have add a chip's test at the beginning and it's good.

pascallanger, 2016-11-30:
I'm not sure which version you are referring to but I can't reproduce any of these errors on the latest source posted on master.
I don't even see how it's possible with the new structure since when you comment CYRF6936_INSTALLED DEVO is not part of the compilation at all anymore. Validate.h is making sure of that...
Pascal

tipouic, 2016-12-01:
Sorry but I wanted to go too fast on a fork to add a WII Nunchck as input and I had my declaration changes after this include.

Now I will try to find how to update this data without slowing down the system. I think I'll interrupt.

closed 2016-12-01 by tipouic