pascallanger/DIY-Multiprotocol-TX-Module #28: Module_BG_4-in-1.md , my module is different?

buzztiaan, 2016-12-05:
http://i.imgur.com/y83AfJm.jpg from a review in banggood suggests the 'V1.2' model thats currently sold by banggood will only need shorting ..

i'm gonna try this, but thought you might like this image to be included in documentation

will report back if it works ;)

buzztiaan, 2016-12-05:
yep, this works :) you might want to add it to the documentation

Dutchy-79, 2016-12-05:
Did you read the guide? It's already in there. Also on rcgroups.com it's in the YouTube video.

buzztiaan, 2016-12-05:
the specific hardware isn't represented by a picture in Module_BG_4-in-1.md ;) 

this was why i opened the issue :)

Dutchy-79, 2016-12-05:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Hardware.md 

First option, it's the exact same thing :)

buzztiaan, 2016-12-05:
its, not, look closely ;) details matter

buzztiaan, 2016-12-05:
most specifically , the -exact- place where you need to add solderbridges (or in some versions, resistors) is quite different

JurikRistjouw, 2016-12-05:
I have also the version 1.2 and just shorted the bridges which looked similar. I did need an USBASP flash programmer as an "Arduino as ISP" did not work. I did not flash a bootloader but just compiled the firmware with the modules i wanted (not commented the hardware protocols as that is also ambigious in the howto's. After that an OpenTX version on the Taranis which supported the serial protocol. Only thing is, the settings of the module are not remembered when saving settings on CompanionX. Otherwise it works like a charm.

buzztiaan, 2016-12-05:
my steps to get the 1.2 working was ;

1) add a 6pin header on the ISP part
2) set protocol rotation knob to 0 (zero)
3) set fuses with : sudo avrdude -B 20 -c usbtiny -e -u -p m328p -v -U lock:w:0x3f:m -U efuse:w:0x05:m -U hfuse:w:0xD3:m -U lfuse:w:0xFF:m
4) flash firmware with : sudo avrdude -b 19200 -c usbtiny -p m328p -v -Uflash:w:Multiprotocol_V1.1.5_CC2500-NRF24L01.hex:i
5) bridge the TX and RX solderpads with a blob of solder
6) done :D

Dutchy-79, 2016-12-05:
https://www.rcgroups.com/forums/showthread.php?2709034-Banggood-Multiprotocol-module-(Flysky-Frsky-Devo-DSM2)

Check the video on post 3

And I have the same 1.2 board and also didn't doubt if I soldered the right pads.

buzztiaan, 2016-12-05:
are you suggesting the video should be embedded with the documentation? my issue was about the documentation -here- 

also, that video is -not- the 1.2 version, note the different text on silkscreen

JurikRistjouw, 2016-12-05:
Can somebody tell me why I should burn the fuses? I did not do that and it worked... I did burn the bootloader (same thing?)

buzztiaan, 2016-12-05:
the fuses determine the options and working speed and -if- a bootloader is used to the atmega chip

its quite possible they were already right for how you use it now

(also the name fuses is weird , they are not fuses, just flashable memory)

JurikRistjouw, 2016-12-05:
Thanx! So in order to install a bootloader (for instance, to update through serial) you need to flash the proper "fuse" in order to make that possible? otherwise, you're stuck with the options on the "fuse" that was shipped?

buzztiaan, 2016-12-05:
yeah and possibly your module already had fuses set to use a bootloader, you wont get stuck (UNLESS you set fuses to disable the ISP (called SPI for fuses) flashing, which would be bad)

pascallanger, 2016-12-05:
Can I have a better picture with no arrows, good light and outdise of the box?
Pascal

buzztiaan, 2016-12-05:
sadly i dont have my camera here :( i'll make you one asap

buzztiaan, 2016-12-05:
some are here;

http://img.banggood.com/customers_images/large/2016112704320712-1048377.jpg
http://img.banggood.com/customers_images/large/2016112306162814-1048377.jpg
http://img.banggood.com/customers_images/large/2016111204035898-1048377.jpg
http://img.banggood.com/customers_images/large/2016102107035467-1048377.jpg
http://img.banggood.com/customers_images/large/2016102107035984-1048377.jpg
http://img.banggood.com/customers_images/large/2016102703094227-1048377.jpg
http://img.banggood.com/customers_images/large/2016102703093771-1048377.jpg

pascallanger, 2016-12-05:
Ok thanks. I'm closing this subjet and add the image.
Please use the forum to talk/exchange/chat...
Pascal

closed 2016-12-05 by pascallanger

buzztiaan, 2016-12-05:
thanks <3

buzztiaan, 2016-12-05:
i didnt use the forum because 438 pages of information is not in any way usable as reference ;)

JurikRistjouw, 2016-12-05:
thanx

gerrievanzyl, 2016-12-05:
Hi all. I am traveling and my data access is sketchy. I will add the image
in the next few days.

buzztiaan, 2016-12-05:
its already added ;)