pascallanger/DIY-Multiprotocol-TX-Module #30: TX ID Question

soligen2010, 2016-12-08:
This is not really and issue, but perhaps question/discussion. . I hope it is OK to post this here, if not please direct me to the right place.

Background: I have built my own transmitter and receivers using my own protocol.  Because I would like to upgrade to an open/tx based transmitter, I want to add my protocol to to this project.  I have this basically working and an testing, but what I would also like to do at some point is release my protocol and receiver code as open source so that we would have an end-to-end solution. Open/Tx, Multi-protocol, and a receiver (using NRF24L01).

Using my protocol, I currently do not re-bind each time I use a model - each receiver is bound only once (to a specific model ID).  For this to work if multiple people at the same field use the protocol, what I need is to set a truly random TX id in the module, saved in EEPROM.  I see a random ID being saved, but it looks like this gets re-created by at least one protocol (HUSBAN, I think it was).  It is also not clear to me if this will truly create a different ID for every module - I could not readily figure out how the random function is seeded.

Can you give me some direction on how to do this that fits with the multi-protocol code? Since EEPROM is already being used, I don't want to step on anything, and prefer to use what you already have if there is something appropriate.

Thanks

pascallanger, 2016-12-08:
The ID generated is different for every module. It's randomely generated at first boot if nothing exist in the eepmrom. Hubsan is used only to force a new random ID generation.
So basically don't think about it, this is how all other protocols are working...
You need to use rx_tx_addr[0..3] in your protocol to access this ID. This is basically all you need to know.
Pascal

soligen2010, 2016-12-08:
Very coo, thanks.  Of course this is all I need to know, but I cant help but be curious on the seed generation.   Based on how the seed is generated, the watch dog timer must have enough variability that gWDT_entropy is different each time.  True?

pascallanger, 2016-12-08:
Yes it does give a lot of variability. It's a well known way to get good random values.
Try it, you'll see that you have a complete new 32bits value at each power up.

closed 2016-12-09 by pascallanger