pascallanger/DIY-Multiprotocol-TX-Module #31: Serial Packet Timeout

soligen2010, 2016-12-10:
I have been struggling with the serial link to the module.  I finally tracked down the issue - even with the correct baud rate, it is  taking longer than the 3250 uS timer for the packet to arrive.  I am pretty sure it is because I am using software serial in my TX (I need the one UART to talk to the computer).

My request is to increase this to 3750.  The snippet below shows the changes. Change the 6500 to 7500

					#if defined STM32_BOARD
						uint16_t OCR1B;
						//OCR1B =TCNT1+(6500L);
						OCR1B =TCNT1+(7500L);
						timer.setCompare(TIMER_CH2,OCR1B);
						timer.attachCompare2Interrupt(ISR_COMPB);
					#else
            //OCR1B = TCNT1+(6500L) ; // Full message should be received within timer of 3250us
            OCR1B = TCNT1+(7500L) ; // Full message should be received within timer of 3750us
						TIFR1 = OCF1B_bm ;		// clear OCR1B match flag
						SET_TIMSK1_OCIE1B ;		// enable interrupt on compare B match
					#endif

Another question I have is the power settings for NRF24L02.  The high power setting doesn't select the highest setting.   I would prefer the highest setting for best range.  Is there a reason to not do this? If not, then I request the high level be changed to:

     #define NRF_HIGH_POWER		NRF_POWER_3

I also I have a question about the bind and autobind flags in the serial protocol.  These are 2 seperate bits, but appear to not be differentiated anyplace in the code as they both turn on autobind.  I was going to use these for different bind behaviors.  Any plans to separate these flags apart?

Thanks

Dennis

pascallanger, 2016-12-10:
1. Serial timing
Short answer: no
Long answer: the current TX (Taranis, 9XR Pro, 9XR...) are using software serial to send the data over the ppm pin in a little less than 3000µs. So 3250µs is already providing a good margin... They are transmitting a packet every 7000µs. To prevent the tail to catch the head I therefore need a timing which is less than half of it. So the maximum value is 3499. I could increase to 3400 but I won't go further...
2. NRF24L01 power setting
Short answer: no
Long answer: have you done some field tests like we did? What I can tell you from my experience is that none of the cheap NRF modules sold out there are working properly at a rf power of 3. They send garbage more than anything else at this power level. From the field tests, it appears that there is no range gain between 2 and 3 but going further there is far more packet loss with power 3 than power 2. This is obvious on some models that just do not run at rf power 3 due to the heavy packet loss.
3. Bind flags
They have different behaviors even if you haven't seen them.
If autobind is set in the serial stream, the module will always start the protocol in bind mode. If autobind is not present but there is a bind request then the protocol will be restarted in bind mode for this time only.
Pascal

soligen2010, 2016-12-10:
Thanks for including the long answers :)

1) I will go through my code and see if something is slowing it down.  Could be my timer interrupt, but the interval is long enough I would have thought I would get a few fast packets through.  8N1 works, it is adding the parity bit that pushes it over the limit.  Also as an not a big deal but FYI, the code comment says 8E2, but the flags configuring the serial look like 8E1 to me.

		//Set frame format to 8 data bits, even parity, 2 stop bits
		UCSR0C = _BV(UPM01)|_BV(USBS0)|_BV(UCSZ01)|_BV(UCSZ00);

Found this bit if doco:
         bit 3	USBS0	USART Stop Bit Select. Set to select 1 stop bit. Unset to select 2 stop bits.

Just FYI.  I am not asking for a change - configuring to 2 stop bits would likely make my timing issue harder to overcome.

2) No I haven't tested this but I will.  I started with max setting and have been very pleased with the performance to date - I have yet to have a brownout flying. Thanks for the info.

3) Ahh - I see it now.  The bind bit re-starts the protocol.  If bind bit stays set the protocol will re-start continuously, restarting the autobind sequence, hence it will stay in bind mode until the bit is turned off and the autobind countdown ends.

soligen2010, 2016-12-10:
I might be wrong on the stop bit thing.  I have found examples both ways :( and the term set/unset is ambiguous.  I would assume set = 1 and unset  = 0, but maybe not.

My understanding is that sending 2 stop bits to a listener for 1 stop bit doesn't really make a difference other than speed.

soligen2010, 2016-12-10:
I optimized the Serial code path and made sure my interrupts don't fire during the send, and I have the serial communication timing looking good.  3040 uS for 8E1 but 3300 for 8E2.  8E2 still works, but I cant tell if it is dropping some packets.  8E2 timing is marginal at best.  but since 8E1 seems to work, i'll stick with that.

pascallanger, 2016-12-10:
Question 1:
UCSR0C = _BV(UPM01)|_BV(USBS0)|_BV(UCSZ01)|_BV(UCSZ00);
This is correct, I don't know where you are looking but the atmega328 datasheet does not leave any place for confusion:
Bit 3 – USBSn: Stop Bit Select . This bit selects the number of stop bits to be inserted by the Transmitter. 0->1 Stop, 1->2 Stop
Question 2:
Multi does care to see 2 stop bits during the transmission as well as the parity being even. If this is found to be wrong anywhere the full frame is discarded.
Question 3:
A stop bit is in fact a pause in the transmission. So if you have a pause between the bytes being sent (computation of the next one for example) this will "emulate" a stop bit.
Like I've said a TX like 9XR Pro is sending a full packet with a true 8e2 format @100K in a little less than 3ms with no packet loss.
Pascal

soligen2010, 2016-12-10:
Thanks Pascal

I just don't have as much processor as the 9XR pro.  They have an ARM and I have a ATMEGA328.  I cant see how I can make it any faster, but my code that captures the timing may add a little, so I'll test some more by playing  with the multi timing and seeing how close it really is.

I have not noticed any issues with 8E1, but I would rather be in spec.  For 8E2 3400 uS is what I would need based on the timing captured from the TX.  The reason I think it actually works may be because there is a little time passes to execute code to start the timer, and capturing the time may increase it slightly.

I know I can always change the code just for myself to get a bit more headroom, it just that when I am done I would rather not have my own one-off changes because it makes getting new versions running more difficult.

pascallanger, 2016-12-11:
Hummm, Multi is also using an atmega 328 and the software serial is working perfectly fine in 8E2 @100K for the telemetry back to the TX. May be you should take it...
Pascal

soligen2010, 2016-12-12:
The issue isn't with the individual bytes, it is the total packet time.  More time between bytes doesn't hurt anything other than total time.  I think if the packet is shorter, or the TX has a more generous timeout, then there would be no issue.  What I did to get the time down was pre-calculate the parity bits for the entire packet before starting to send.  The fact that 8E1 works I figure means that my stop bit timing is likely still a little long as it cycles to the next byte, even though I made this as fast as a could with reasonably structured code.  There is still a function call for each byte - oddly making the function inline slowed things down.  I would have expected it to be faster.  Perhaps due to register optimization.  

I would be interesting for you to try 8E1 back to the TX, but perhaps not worth your time to play with it - just a curiosity.

pascallanger, 2016-12-13:
Closing this none issue ;-)

closed 2016-12-13 by pascallanger