pascallanger/DIY-Multiprotocol-TX-Module #32: Unable to control WLToys A959 after binding.

KevSp, 2016-12-10:
Hi, I know this was raised and closed but I believe that was an issue on the user side.

I have tried today to connect the following configuration but have no control after binding;

Hardware: Tx Turnigy 9XR PRO [Mod REVB-X]
Firmware: ersky9x-r220 [04.12.2016]
Module: 4in1 Multiprotocol [Banggood]
- Tx/Rx bridged for Serial config
- Flashed via USBasp with download from 02.Dec.2016

JJRC Branded WLToys A959 [brushed]

I get a bind light on both the Tx and Rx but no response to inputs on the Tx.

Tx is set for:
- External Module
- Proto: Multi
- RxNum: 0
- Protocol: V2x2
- Type: None
- Autobind: N (have tried Y)
- Option: 0
- Power: High
- Rate: 7ms
- 1st Chan: 1

Hope this helps with any diagnostic - hope it's just an ID10T issue on my part.

Cheers

KevSp, 2016-12-10:
My apologies - should read A595 RC, read my model setting incorrectly as I also have an F959 on here (which does work).

KevSp, 2016-12-10:
So the bind seems okay and have sorted that there's no default mixer.

For Mode2 Throttle seems to be on Ch2, Aileron on Ch4.

But have a problem going from forward to reverse - will hit reverse once after throttle enable and not again after going forward.

Have checked with original controller and it functions - is this just me not getting a setting right in Mixer or something in the protocol?

pascallanger, 2016-12-10:
What are the mixers on your 9XR Pro?
CH1=A, CH2=E, CH3=T, CH4=R? At least this is what multi is expecting by default unless you've changed it in _config.h file. Mode 2 or Mode x has no influence here.
Pascal

pascallanger, 2016-12-13:
I haven't heard back from you... Is it working, not working?
Have you tried the latest version of the V2x2 protocol?

KevSp, 2016-12-14:
I'm back home tomorrow morning so will update everything and check again - any mixers are default though but updating the V2x2 sounds like a possibility.

pascallanger, 2017-01-01:
Since I've not heard from you for the last 18 days, I'm closing this issue...
Pascal

closed 2017-01-01 by pascallanger