pascallanger/DIY-Multiprotocol-TX-Module #38: add support for this module to openground

jacobbloy, 2017-01-20:
is there any way we can add support for your module to this firmware  for the turnigy i6s and turnigy evolution. https://github.com/fishpepper/OpenGround

pascallanger, 2017-01-20:
Hi, I'm not sure what you are looking for bt here is an attempt.
The Multiprotocol module uses a serial interface to communicate with the TX. It receives data from the TX which contains protocol, sub_protocols, other details and stick data and sends telemetry data if the protocol supports it.
The serial protocol used is explained in details in the multiprotocol.h file line 417 and up: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multiprotocol.h
Let me know if you need more information.
Pascal

jacobbloy, 2017-01-20:
I will have a look and see if my skill level is high enough. In serial mode does the DIY Modual convert the TX data like stick position into a ppm signal or is it already expecting this? As the transmitter does not already have a ppm out so I guess my question is, will I have to write a ppm timer?

pascallanger, 2017-01-20:
Again I'm not sure what you are looking for...
If you want to connect the transmitter using PPM then just use the PPM entry and use the dial switch as explained in the documentation to select the protocol you use.
Now if you want to integrate the module functionnality in the radio GUI then you need to use the serial mode. I thought this is what you were asking for since the first use case is clearly explained in the documentation of this github.
Pascal

jacobbloy, 2017-01-20:
What is your hourly rate to implement it for me? jacob.bloy@gmail.com

pascallanger, 2017-01-20:
I still don't understand what you are looking for. Why don't you ask fishpepper since it's his project?

jacobbloy, 2017-01-20:
I'm looking for help. The skill level needed is a bit out of my depth and fishpepper is currently working on his ESC and FC project and doesn't have enough time.

andremelin, 2017-02-01:
we tried it today with latest release on both sides rx and tx opentx2-2-362 with DMS X and telemetry with the Spektrum T1000 hub on the Diy Multiprotocol of Banggood with . Yet I have some questions. I measured the RF output power of 22DBm. On the Taranis Plus the telemetry recognised the Alt and Vspd and some other "F de A...b...c and so on". Range mode test was about 80meters. In flight we got RF signal critical message at 100m hight and 60m horizontal distance so around distance real of 128m only. Is this normal ?How comes we get the RF Critical message, is it to do with the opentx RSSI value ? How did you get to define this value and where ? Thanks for explanation in advance because with 22DBm rf signal output power which is largely above the 20DBm max power range of normal emitters we can consider being long range and over that???

jacobbloy, 2017-02-09:
can you please point me to the right file where stick and switch positions are sent via RX/TX. i have got the protocol switch side done.

pascallanger, 2017-02-09:
Already answered on the 1st comment...

closed 2017-02-15 by pascallanger