pascallanger/DIY-Multiprotocol-TX-Module #45: V1.1.6 telemetry no longer working with openTX 2.2

benoit35690, 2017-02-02:
Test performed with 
- multiprotocol module V1.2 from Banggod
- Taranis X9D
- openTx 2.2 RC10

see openTx issu 4373
https://github.com/opentx/opentx/issues/4373

pascallanger, 2017-02-02:
I'm pretty sure the problem does not come from Multi. Kilrah has tested and reported it working and same for schwabe on your original bug report.
Your _Config.h is old but correct.
At this stage you might have a hardware issue. You need to verify with a multimeter that you have an actual good connection between the 5th pin of the radio connection (the bottom one) and the atmega TX pin.
Pascal

pascallanger, 2017-02-03:
Humm if the problem is a bad soldering on one of the 2 bridges then it's not their fault...

closed 2017-02-03 by pascallanger

benoit35690, 2017-02-03:
I have double check the 2 solderings on the TX and RX bridges

pascallanger, 2017-02-03:
And the continuity from pin 5 to Atmega.tx?

benoit35690, 2017-02-03:
sorry my skill in electronic is not enough for that test

benoit35690, 2017-02-08:
Hello Pascal, could you please explain me where is PIN5 ?
I should check the continuity with the TX pad (where the soldering should be present) ?
I saw the issue you see when all module are not enabled, in my case they are all 4 enabled.
I will be able to do some more testing in a few weeks since I go on business trip.

pascallanger, 2017-02-08:
Please use the forum to talk about this issue: https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page521#post36832945.
I can't post any pictures here.
Others can help as well when I'm not around.