pascallanger/DIY-Multiprotocol-TX-Module #48: X9D+:4-in-1:Multi 1.1.6.13 - Sticks on No Multi_Telemetry

powercycle, 2017-02-06:
Setup: radiio has multimodule 1.1.6.13 with Q303 as a compiled protocol (sub CX10WD)

How to see problem:
Set protocol to Q303, status shows module version.
Cycle through protocols until No Multi_Telemetry is found.
Cycle back to Q303, No Multi_Telemetry still displayed.

How I fix it:
Power cycle the radio on the Q303 as the selected protocol.

Also a ticket for OpenTX is open.  https://github.com/opentx/opentx/issues/4337

pascallanger, 2017-02-06:
I can't reproduce the issue on my 9XR Pro running ersky9x.
Could you try the latest github release?
Could you attach your _config.h file?
Pascal

powercycle, 2017-02-07:
Tried 1.1.6.15, no change.  Fails on Esky protocol.
Interesting to note that it no longer responds to protocol change.  Red light is on and stays on even on protocols where it would have gone into binding mode before.  Item on menu does change, just not status.
[_Config.h.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/756364/_Config.h.txt)

pascallanger, 2017-02-07:
Issue solved on 1.1.6.16. But in any case read the comments for the configuration of the module, for the 4in1 all modules are installed so the CC2500 should not be commented...

closed 2017-02-07 by pascallanger

powercycle, 2017-02-08:
testing 1.1.6.16, works.  Additionally corrected config file to include all the modules like the comments in _Config.h say to do.  Duh.