pascallanger/DIY-Multiprotocol-TX-Module #50: 4 in 1 multi protocol, Taranis x9e and opentx 2.19, hubsan protocol and telemetry

dprerry20, 2017-02-10:
I installed the 4 in 1 module in my taranis x9e. Upgraded firmware on transmitter and modified the 4in1 muti protocol board from bangood version 1.2. 
I can get the transmitter to bind with the hubsan h502s
I have no telemetry even though it was enabled in the config.h file. 
The only way I can get the quad to arm is by taking it outside till it finds at least 6 satelites.
I have no way of over riding the fly with no gps.
Any way of being able to get these function on the ex9

Thank you guys for all your work

pascallanger, 2017-02-11:
Hubsan protocol does not support all this yet. This is a future improvement.

dprerry20, 2017-02-11:
> <DIY-Multiprotocol-TX-Module@noreply.github.com>
> <author@noreply.github.com>
> protocol, Taranis x9e and opentx 2.19, hubsan protocol and telemetry
> (#50)

Thank you

pascallanger, 2017-02-15:
I'll look at the pull request for hubsan when I'm be back from vacation. Stay tune.
Pascal

closed 2017-02-15 by pascallanger

dperry20, 2017-02-15:
I tried the code yesterday. It didn't do anything. I am not a 
programmer. I did some programing in fortran and basic in my earlier 
days. There were a couple of errors which I think I fixed because when I 
ran the sketch, no further errors showed up.

I uploaded the sketch to my 4 in 1. Nothing showed up. Is there anyone 
that can check the code that I uploaded for the Hubsan.

Danny

Cc: "dperry20" <danperry52@netscape.net>; "Manual" 
<manual@noreply.github.com>
Sent: 2/15/2017 9:00:43 AM

pascallanger, 2017-02-15:
Like I said I'll look at it when I'm back...

dperry20, 2017-03-14:
It did bind.