pascallanger/DIY-Multiprotocol-TX-Module #53: Can't modify the channel order.

famtory, 2017-02-22:
I used google translate.
==================
//Modify the channel order based on your TX: AETR, TAER, RETA...
//Examples: Flysky & DEVO is AETR, JR/Spektrum radio is TAER, Multiplex is AERT...
//Default is AETR.
#define AETR
=================
I changed AETR or TAER or more.
But same result. Nothing changes.
why?

pascallanger, 2017-02-22:
What do you mean? I've tested it TAER and it works fine.
If the module is set to TAER and the radio is sending TAER then the modules converts it to whatever the protocol needs. What I mean it's normal that even if you put TAER and send TAER, the output on a flysky receiver will be AETR.

famtory, 2017-02-22:
I have JR 10x.
It is TAER.
but I want AETR.
.
======================================
               ppm
JR(TAER) -----> multiprotocol tx -> receiver_ppm(AETR)
=========================================
It is possible?

pascallanger, 2017-02-22:
AETR, TAER, RETA,... is the module input only.
The output order is forced by most if not all protocols you are selecting.
Which protocol are you talking about?

famtory, 2017-02-22:
FLYSKY 2A - TAER
DSMX - TAER
FRSKY - TAER
Regardless of channer order(AETR or TAER), always TAER

pascallanger, 2017-02-22:
Why would you want that?
You can do it if you want by modifying the code...

famtory, 2017-02-22:
I have futaba 1024z with frsky module.
and Jr 10x with multiprotocol.
I usually use Futaba, but sometimes I use JR.
I set the switch to the same position as the flight mode and Arming.
BUT, the channel order was different, so I had to set it again.

pascallanger, 2017-02-22:
This is an edge case that I won't address as this will complexify the configuration and will loose people even more... You can modify easily the protocols yourself to do what you want if you wish.
Pascal

closed 2017-02-22 by pascallanger