pascallanger/DIY-Multiprotocol-TX-Module #56: Bayang telemetry not binding

Dakkaron, 2017-02-28:
Hey,
When I try to bind without telemetry enabled, everything works fine, but when I try to bind with telemetry enabled (option value==1), the binding is just not established.

My setup this:

- Eachine H8 mini Blue Board Version
  - running the software from https://github.com/silver13/H8mini_blue_board/
  - I am using the version newest version (last commit on 19.02.2017)

- Self-built Multiprotocol module with software from here
  - The version I am using is based on the last commit from 24.02.2017

- Taranis X9D+
  - Version 2.2.0N362

I tested my H8 with a friend's Devo 7e and there telemetry works fine. So I guess, it is not a problem with the H8 firmware.

For testing purposes I modified the Multiprotocol Module firmware to have the option value hardcoded to 1. This caused the same behaviour as just setting the option value to 1 (cannot bind). I also tried to change the settings on the Taranis to either "Protocol Bayang Bayang, Telemetry 1" or "Protocol Custom 14 0, Option 1, which should be equivalent and the behaviour is the same. If I set Telemetry/Option to 0, I can bind but have not telemetry, and if I set it to 1 I cannot bind.
From this I conclude that the problem is not the Taranis software.

So the last thing that's left is the Multiprotocol Module. What can I do to get this working? Is it working correctly for other people?

pascallanger, 2017-02-28:
It works fine for me on a H101 using https://github.com/silver13/H101-dual .
I can bind it using both modes and get telemetry working with option=1.
When you say that the quad does not bind, what's the led status of the module and the quad? Do you see it trying to bind?
Pascal

Dakkaron, 2017-02-28:
Good to hear that it can work at all, so it is probably a problem with my setup only. My module has no bind LED, but the Taranis indicates it is binding for a second or so. The H8 does not indicate anything regarding binding.

Is there anything I can do to diagnose the problem better?

Thank you for the quick reply!

pascallanger, 2017-02-28:
Summary:
 - H8 mini Blue Board works fine with telemetry on Devo 7e
 - H8 mini Blue Board works fine with Multi with telemetry turned off
 - H8 mini Blue Board does not bind with Multi if telemetry is turned on
 - H101 board works fine with Multi with  and without telemetry turned on
This is weird... There must be a difference between the 2 SilverX implementations causing this issue... I know the H8 version is working as well since I got reports on the forum.

Let's remove the switch to RX to see if it helps binding, can you comment this part of the bayang protocol:
```
	#ifdef BAYANG_HUB_TELEMETRY
    if (option)
	{	// switch radio to rx as soon as packet is sent
 		while (!(NRF24L01_ReadReg(NRF24L01_07_STATUS) & _BV(NRF24L01_07_TX_DS)));
		NRF24L01_WriteReg(NRF24L01_00_CONFIG, 0x03);
    }
	#endif
```
Pascal

Dakkaron, 2017-02-28:
I am sorry, it was a false alarm. The software is working perfectly fine, but I messed up when building the module. I accidentally soldered MISO to IRQ on the NRF24L01. This was apparently not a problem for just sending, but it was a problem if I wanted to receive as well. Thanks a lot for your help. I had this problem ever since I first built that multiprotocol module. I must have checked for problems in my circuit at least 15 times, but I did not know that it was just a problem on my module's hardware or if other parts of the system were involved. Your assurance that the software itself works really helped me. I don't know if I would have found the problem otherwise.

Now binding and telemetry work fine. Thanks a lot for your help and the great work with this software!

pascallanger, 2017-02-28:
Humm I lost nearly 1 hour looking at this issue and trying to find out what was the difference between the h101 and h8...
If you want to support my work and the project you could donate something.
Pascal

closed 2017-02-28 by pascallanger

Dakkaron, 2017-02-28:
I am really sorry for this, and I will donate something for this.

pascallanger, 2017-02-28:
Thanks, really appreciated !