pascallanger/DIY-Multiprotocol-TX-Module #58: New Protocol

soligen2010, 2017-03-11:
I don't know if you recall, but a few months ago I mentioned that I am working on an open source protocol and I wrote the code for Multiprotocol to be the transmit module.  I have finished up the receiver portion, and I was hoping to contribute the xmit protocol here.  I wanted to post first to get guidance to be sure you are OK with this, and how you would like me to proceed.  I'll need to re-merge my changes as you have made updates since I last forked the code,

If you want to take a look, the Xmit code is in the soligen2010 branch of Multi. [https://github.com/soligen2010/DIY-Multiprotocol-TX-Module](https://github.com/soligen2010/DIY-Multiprotocol-TX-Module)

Receiver code is at [https://github.com/soligen2010/RC_RX_CABELL_V3_FHSS](https://github.com/soligen2010/RC_RX_CABELL_V3_FHSS)

Thanks

pascallanger, 2017-03-12:
Hi,
Are you open for improvements?
I think adding telemetry to the protocol is a must. Nowadays people are at least expecting to get the link quality (RSSI or other mean), voltage from the flight battery... You can have a look at the Bayang protocol which includes all that: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Bayang_nrf24l01.ino and the RX implementation here: https://github.com/silver13/H101-dual/blob/master/H101_dual/src/rx_bayang_protocol.c
Pascal

soligen2010, 2017-03-14:
Yes I agree.  The 2 new features I think this needs are telemetry and serial (SBUS) output.  Since I currently don't have any SBUS or telemetry gear I wanted to get this out there an see what support/interest I get,  In the mean time I want to do some range testing and build some new planes.  I have had good luck with MAX power settings, but I want to test it better.

I updated from your master branch and i can make a pull request.  The update process created a big commit that it looks like will show up in the list of commits on the pull request, but it is just all the things you have done since I made my fork.

Is it OK to submit the pull request?

soligen2010, 2017-03-28:
Hi,

I have been looking at telemetry.  I have return telemetry packet transmitting back a stub packet.  I have looked at Bayang and it all looks good, but I have some questions.

What is compensated vs. uncompensated lipo voltage?

Can you explain what the telemetry_lost and telemetry_link do?  I think I figured some of it out, but not sure.  I know I can simply replicate what Bayang does, but I prefer to understand,

Is there a way to output the amps if I can return that from the RX?

What is RX and TX LQI?

What is the scale for RSSI.  Looks like packets per second, which on Bayang would cap out at 200 (every 5 ms if I read it right).  I hope to keep a 3ms timing (some initial tests show it is possible) so my packets per second will exceed 255.  Should I re-scale to 100% = 200 or 255?

Is there anything I have to change in any file other than the CABELL protocol file?  How to I tell the rest Multi that this protocol should expect to return a telemetry packet?

soligen2010, 2017-04-18:
I have some basic telemetry working.  RSSI is successfully coming back to the module. I made the RSSI scale 0 to 255.
Any info on compensated vs. uncompensated lipo voltage or amps? I haven't done anything with this as I haven't decided what hardware to use to.

Thanks

schwabe, 2017-04-18:
I think the compensated vs uncompasted lipo is something specific to that protocol. Most protocols will just have voltage and ampere.

soligen2010, 2017-04-19:
That could be. The thing is that I don.t see where I would put the amps if I start sending amps back, so I was wondering if it has something to so with that.  I am getting the packet back from RX and see what variables to put the data into, but I am pretty unclear on how the packet gets formed to go back to the TX.  Not may comments in the code, and I haven't unraveled the basic premise on how it works yet. Any info that would help me understand would be appreciated.

soligen2010, 2017-04-19:
Looked at it a little more.  Looks like I need to change frsky_link_frame() to include my protocol in the IF statement.  Should this be all I need to do to get the telemetry to come out?

Looks like I should v_lipo1 should be all I should worry about, but I see a comment about it being volts*100/2.  IS this value then only the single cell voltage and not the whole pack voltage? (if so, so the TX get programmed with this cell count to do the multiplication?)

Thanks for any help understanding this.

KnuckleUpFPV, 2017-10-17:
Only flysky receiver that does sbus is the fs82a. The ibus protocol native to flysky is actually faster than sbus.

lostinwoods, 2017-11-06:
Did you test it yet? Such protocol is the only thing nrf24l needs to replace whole zoo of modules.

KnuckleUpFPV, 2017-11-06:
I just got a notification from where i commented on this before. My comment is in the wrong place. I had too many windows open and commented here on accident. Please remove both of my comments as they serve 0 purpose to this project.

soligen2010, 2017-11-07:
Yes I have been using this quite a lot.  I flew this protocol at Flite Fest East with lots of other planes in the air.  I did have one brown out cause a crash, but that was during a combat with over 100 planes in the air and that particular receiver was older hardware and didn't have diversity.

I think the software is solid and I am working on hardware.  The telemetry signal seems to have range issues so I have been playing with antennas, and I think next step is try better power filtering as my latest theory is that servo/motor noise is affecting the RX in the plan when it transmits.  With telemetry turned off I don't notice any issues.  As I am using PA/LNA modules on both RX and TX it seems odd that telemetry should have more problems than the control signal.

lostinwoods, 2017-11-14:
Your project is really interesting! The one I was trying to find so long. One question - is it possible to control power of the module, to limit it for indoor model`s? And the second thing is request of unlimited version - nrf24l works in range 2.4000 - 2.4835 Ghz. In other countries regulations are different from ones in US...

soligen2010, 2017-11-14:
TX Power is controlled by Multiprotocol just like any other protocol.  It has setting for HIGH and LOW power .  This new protocol also has a bit in the option byte that will convert HIGH power to MAX power, so with thus protocol there are 3 possible settings, LOW , HIGH, and MAX.

On the RX side for telemetry the power will be MAX if the TX power is MAX, otherwise ti is HIGH.  There is currently not a way to make the RX send telemetry using LOW power, so if HIGH to too much to be legal for you, don't use telemetry.  

The protocol work son frequencies 2.403 to 2.448.  This is in the HAM radio band as well as the typical ISM band.  Changing this is easy, but does require you to re-compile both the TWX module and the RX module.  There are 2 lines that need to be changes to adjust the frequency range used:

#define CABELL_RADIO_CHANNELS         9                   // This is 1/5 of the total number of radio channels used for FHSS
#define CABELL_RADIO_MIN_CHANNEL_NUM  3                   // Channel 0 is right on the boarder of allowed frequency range, so move up to avoid bleeding over

Keep CABELL_RADIO_CHANNELS under 20 or else some variables will overflow producing unpredictable results.

What are the constraints in your country?  I highly recommend getting a HAM license as this may reduce the constraints.

lostinwoods, 2017-11-14:
My country is to wild to care about licenses. Nobody really care, the reason I am talking about possible changes is that city is too polluted by 2.4 radio signals and to find less populated  frequency I have simple arduino based scanner.

soligen2010, 2017-11-14:
Rx_Tx_Util is responsible for setting the channel sequence. and returning the next channel.  You can either use the #defines above to tweak the range of channels used, or replace the function with whatever you like - even hard coding the channel sequence if you so desire.  You just need to do the same thing in the Multiprotocol code because both sides need to know channel sequence.

That said, the hopping being done is pretty resilient.  It took over 100 other planes in the air at the same time before I lost control due to signal without it being a range issue.  And, that only happened once.  The other time I flew a combat with almost just as many planes I didn't have an issue.  And both of those were on older non-diversity receivers.  The diversity receivers work even better.

Since packets are sent every 3 ms you can drop many packets and still have good control over the model.  Even if you drop 10 in a row, you only add 30ms to the latency, which I think is still very flyable in most situations.

lostinwoods, 2017-11-15:
Now the only question for me is why it is still not merged) 
You did a great job!

soligen2010, 2017-11-16:
Thanks.  I simply haven't yet pursued getting it merged.  I was hoping to first resolve why telemetry range is so poor.  I suspect it is hardware, but the reason is unclear.  I had a problem with a batch of bad arduinos, which caused issues that prevented any good trouble shooting.  Now that the bad arduinos are in the trash, and winter is almost here, hopefully I can look seriously at telemetry

lostinwoods, 2017-11-16:
I think everything is much more simple, than you think, for sausage antenna positioning is critical. If it is in horizontal position and points in your direction, you louse signal faster, than when it is in vertical position.

soligen2010, 2017-11-17:
Yes, I get the orientation thing, that's why I added radio diversity to the RX.  Its just that I would think that the uplink (control signal) and downlink (telemetry signal) would be similar in reliability, but they are not. My current theory is noise is getting through my 3.3v regualtor (1117, presumably a  clone) on the RX and affecting the telemetry transmission. Just a theory for now based on observation that when flying the telemetry range seems worse than when I range test by walking up my street.  It's kinda hard to diagnose problems while flying, so I need to set something up on the bench to simulate flying - of course its hard to test range while in the basement on my bench.  Its just really difficult to diagnose since I can't put an oscilloscope on my plane :(.

lostinwoods, 2017-11-17:
Are your PA/LNA modules of [this](https://ru.aliexpress.com/item/1sets-Special-promotions-1100-meter-long-distance-NRF24L01-PA-LNA-wireless-modules-with-antenna/32341792715.html?spm=a2g0s.9042311.0.0.VL9qzN) kind or they have [skreen protection](https://www.aliexpress.com/af/E01%25252dML01DP5.html?SearchText=E01-ML01DP5&ltype=wholesale&d=y&tc=main&blanktest=0&initiative_id=SB_20171117095356&origin=n&catId=0&isViewCP=y)? Engines EM noise influence first type more...

soligen2010, 2017-11-23:
They are the first type, and I add some shielding using copper tape tied to ground sandwiched between 2 layers of heat shrink

pascallanger, 2017-11-23:
I think pictures would help :-)

soligen2010, 2017-11-23:
All my receivers are currently inside models - hard to photograph.  Next time I build one i'll have to do step by step photos.

What I did is similar to this:

[http://blog.blackoise.de/2016/02/fixing-your-cheap-nrf24l01-palna-module/](http://blog.blackoise.de/2016/02/fixing-your-cheap-nrf24l01-palna-module/)

but I used heat shrink instead of plastic wrap, and copper foil tape instead of aluminum foil.  My ground connection from the copper tape is a wire soldered to both tape and a ground connection on the board.

I also remove the SMA connector and directly solder the coax to the board.

The module is powered by an AMS1117 3.3V LDO. Since the output capacitor for this regulator cant be ceramic (ESR must be above .3 ohms) I think the NRF24L01 module is not as well de-coupled as it could be.

My next step is to try a low pass filter after the regulator so I can put more de-coupling caps on the power input to the NRF24L01.  If this ends up helping, I'll abandon the AMS1117 for something better so I can decouple it without adding a low pass filter.

pascallanger, 2017-11-23:
I've added your protocol to my repo.
I had to modify the code to get rid of warnings... I hope nothing is broken.
It would be great if you could add some more details in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md describing your protocol but that's up to you.

soligen2010, 2017-11-23:
Wow, Thanks.  I would have re-merged the code and done a pull request.  Can you recall the warnings?  In Arduino  1.8.0 I don't get any warnings.  I'll do a diff an see what you changed.

pascallanger, 2017-11-23:
In this part of the code:
```
static uint8_t __attribute__((unused)) CABELL_getNextChannel (uint8_t seqArray[], uint8_t seqArraySize, uint8_t prevChannel)
{
	prevChannel -= CABELL_RADIO_MIN_CHANNEL_NUM;				// Subtract CABELL_RADIO_MIN_CHANNEL_NUM because it was added to the return value
	if(prevChannel>(seqArraySize * 5))
		prevChannel=seqArraySize * 5;
```
There was a constrain which I've changed by a if to get rid of the warning. Can't remember exactly the message but it was complaining that the comparaison was always false.
I've got all levels of warnings enabled in the preferences of IDE v 1.8.4.

soligen2010, 2017-11-23:
I don't see a problem with that change.  Theoretically it SHOULD always be false, but I want to be safe as I didn't want to take chances with the channel number causing unpredictable results and cause an out-of-band transmission.  The NRF24L01 is capable of transmitting on frequencies outside the ISM band.

The only other change I noted that was not formatting/style was changing a variable i from uint64_t to a uint8_t in CABELL_getChannelSequence.  In this looks like an oversight because the variable never exceeds 255, but I had done this intentionally since it is involved in 64 bit math. I didn't want to see any oddities happen with the order of operations and when data type conversion happen.  It is my practice to simply avoid data type conversions if I have any doubts.  Testing will tell, but this is the only thing I noted that might break something.

Once I get set up and with your version I will send in a pull request to change

		numChannelsFactorial *= i;      //  Calculate n!

to

		numChannelsFactorial *= (uint64_t)i;      //  Calculate n!

If for nothing other than my peace of mind.

And, yes, I will write something for the documentation and include a link to the RX project.

pascallanger, 2017-11-23:
The link to the RX project is already there... It's more of you want to add other details.
Calculation for nothing on 64bits for a 8bit processor is quite time expensive...

soligen2010, 2017-11-24:
The 64 bit operations only happens once during the protocol init, so the extra time is really not critical.  Once the channel order is established, the 64 bit operations are done.   Since it only happens once, I didn't feel I needed to find a different algorithm for setting the channel sequence that avoids the big factorials.

soligen2010, 2017-11-24:
I think we have something wrong. I synced my fork back to your master and although the protocol seems to work fine, reliability is really low. I think the serial link between the module and the Taranis is the problem.

I think there has to be something in the software causing this or making something worse because I re flashed my old fork and the behavior went back to what I expected.  Here are the symptoms I am observing. These issue happen at short range which previously has been rock solid, and these don't happen on the previous firmware.  Previous firmware was based on 1.1.6. 20, and of course the new is based on your current code.

A lot of telemetry lost messages.
A lot of RSSI low/critical messages
Receiver fail-safeing.
RSSI on the TX would jump to very low values, but monitoring the packet rates on the RX (I have an LCD hooked up to it) RSSI should not be so low
The status field on the taranis changes fairly often between V1.1.6.29, V 1.1.0.0, No Multi-Telemetry 

The status changing and bogus RSSI numbers is what leads be to think it is the serial link.

I have replaced the connector on my module thinking it may be getting worn, and also re-soldered the pins in the Taranis in case it was a cold joint, but this hasn't helped.

Any thoughts/help?

soligen2010, 2017-11-24:
A few more details, the status also changes to some other random version numbers, and also has dis played "Not in Serial Mode".  For completeness, here is _myConfig.h:

#define INVERT_TELEMETRY
#undef MULTI_STATUS
#define MULTI_TELEMETRY
#undef ENABLE_PPM

#if not defined STM32_BOARD

  #undef A7105_INSTALLED
  #undef CYRF6936_INSTALLED
  #undef CC2500_INSTALLED
  
  #undef BAYANG_NRF24L01_INO
  #undef CG023_NRF24L01_INO
  #undef CX10_NRF24L01_INO   // Include Q2X2 protocol
  #undef ESKY_NRF24L01_INO
  #undef HISKY_NRF24L01_INO
  #undef KN_NRF24L01_INO
  #undef SLT_NRF24L01_INO
  #undef SYMAX_NRF24L01_INO
  #undef V2X2_NRF24L01_INO
  #undef YD717_NRF24L01_INO
  #undef MT99XX_NRF24L01_INO
  #undef SHENQI_NRF24L01_INO
  #undef FY326_NRF24L01_INO
  #undef FQ777_NRF24L01_INO
  #undef ASSAN_NRF24L01_INO
  #undef HONTAI_NRF24L01_INO
  #undef Q303_NRF24L01_INO
  #undef GW008_NRF24L01_INO

#endif

pascallanger, 2017-11-24:
AVR or STM32?
I thik the big thing that has been introduced is CHECK_FOR_BOOTLOADER. Is it enabled?

pascallanger, 2017-11-24:
Could you also try to disable:
 - MULTI_TELEMETRY and test
 - TELEMETRY and test
I need to have a clue of where things go wrong...

Can you also tell me if it's only with your protocol or other protocols have the same behavior?

- Pascal

pascallanger, 2017-11-24:
Could you send me a link to the working version of the code so I can do a diff to see if I can spot something?

pascallanger, 2017-11-24:
Ok I think I've found something.
If I'm not wrong, you are on an atmega328p platform using invert serial therefore bitbang serial for telemetry.
When they've increased the buffer to 192 instead of 128, they've updated only half of the code so sometime the buffer is getting corrupted and bad things start to happen like what you are describing.
To test the fix, at the end of Telemetry.ino, you need to replace this single line of code: ```ptr->tail = ( ptr->tail + 2 ) & 0x7F ;```
by this code:
```
				uint8_t nextTail = ptr->tail + 2 ;
				if ( nextTail > TXBUFFER_SIZE )
					nextTail = 0 ;
				ptr->tail = nextTail ;
```
Please let me know if this solve your issue.

Pascal

soligen2010, 2017-11-24:
It is definately the bit banging. What I did to narrow it down is wire in a hardware inverter and turn off INVERT_TELEMETRY and this seems to have fixed it.

To answer your earlier questions, yes atmega328p 16 MHz.  I did obverse the status line issue on another protocol, but it corrupts much less frequently.  When I configure the protocol to not have a telemetry link, things seemed fine until I enabled the telemetry on the RX, then things went bad.  When I went back to confirm the issue on the other protocol I only got the other protocol to go bad after I first had a telemetry link on my protocol.  All these observations confirms your theory about things getting corrupt.  I think that as long as there is no telemetry link then little enough data is going from the module to the tx that all is ok, but once the data flow increases with a telemetry link, things go bad.

I had noticed the buffer size change and tried changing it back to 128, but that didn't fix it. (although I think I misses one of the hard coded 192's when I tries this.

BTW, I also noticed in Telemetry.ino a hard coded 192 in a couple places that I think should probably be TXBUFFER_SIZE instead of hard coded.

I'll have to remove the hardware inverter to test your software fix.  I'll get back to you.

soligen2010, 2017-11-25:
I tried your fix but unfortunately it didn't correct the problem.  FOR now I went back to the hardware inverter which works well.

Just for full disclosure, I neglected to put the 470 ohm resistor on the telemetry line (it was never there).  It doesn't seem like this would be the issue, but if you think it could be the culprit I can put one in.  It's just feeling like software at this point.

soligen2010, 2017-11-25:
I put in the 470 ohm resistor just to be sure.  No change.

soligen2010, 2017-11-25:
Reverting back to a buffer size of 128 and things work fine, so it is certainly something to do with the increase in buffer.  I had tried this before but I missed changing one line.  This time it worked.

Here are some things I noted along the way

I found this line and tried the same fix as what you had done, but it didn't correct the problem.  Perhaps there is even another place then needs to be updated, or this needs slightly different logic.

	uint8_t next = (SerialControl.head + 2) & 0x7f ;

I found these lines with a hardcoded 192.  I replaced 192 with TXBUFFER_SIZE which made it easier to change between 192 and 128

			t += 192 - h ;
.
.
.
			if ( nextTail > 192 )

Just curious, why was the buffer increased in size?  I am wondering if 192 is simply too much for the atmega328p, or if there is yet another piece of code that needs to be changed.

soligen2010, 2017-11-25:
Ahaa.  I got it.

In telemetry, change this one line

uint8_t next = (SerialControl.head + 2) & 0x7f ;

to

  uint8_t next = (SerialControl.head + 2);
  if (next >= TXBUFFER_SIZE)
    next -= TXBUFFER_SIZE;

soligen2010, 2017-11-27:
I saw your latest changes and still an issue.  I put in pull request #113 which I think is the final fix.  All the compares needed to be >= instead of >.  I thought just = would have worked but it didn't.

pascallanger, 2017-11-27:
It's already in... Where do you see it missing?

pascallanger, 2017-11-27:
I don't see anything missing: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Telemetry.ino
Let me know what I'm missing...

soligen2010, 2017-11-27:
The pull request I put in is still open, so I assumed it wasn't merged, but I see you must have made the changes yourself.  Do you prefer to put thing in yourself manually instead of having pull requests?

Just curious for any future contributions how you prefer to get them.

pascallanger, 2017-11-27:
PR is fine, we just found the issue nearly at the same time. So for me it's there but I don't know why the pull request do not show it... It might be because it was in my local repo and as usual I've forgotten to press the push button and your PR came in the middle.

soligen2010, 2017-11-28:
Now that the merge is done and the BASH Serial stuff is fixed, I think we can close this issue.  There is an RC Groups thread for this protocol at 

[https://www.rcgroups.com/forums/showthread.php?2908017-Open-Source-RC-Protocol-Receiver-Project](https://www.rcgroups.com/forums/showthread.php?2908017-Open-Source-RC-Protocol-Receiver-Project)

There is also the receiver project on github which can be used for rx issues at 

[https://github.com/soligen2010/RC_RX_CABELL_V3_FHSS](https://github.com/soligen2010/RC_RX_CABELL_V3_FHSS)

Please feel free to join contribute to the thread and/or the RX project.

closed 2017-11-28 by soligen2010