pascallanger/DIY-Multiprotocol-TX-Module #59: FrSky V8 mode damages receivers?

jmpblackfoot, 2017-03-19:
Bangood Multiprotocol module updated to 1.1.6.20 in a Taranis X7
I bound a FrSky V8R4 in FrSky V8 mode and now the receiver does not work anymore.
Tried an 8-channel early V8 receiver with the same result.
They don't work anymore.
It is as though the binding to the Multiprotocol module in V8 mode altered the firmware in the V8 receivers.
Other D or X-series receivers bind and work fine with the module in D8 and D16 modes.

Isaac96, 2017-03-19:
Invalid TX GUID?
(I think that's what its called)
What happens if you rebind with a different module?

jmpblackfoot, 2017-03-19:
The 4-channel receiver just doesn't operate any more, no LED.
When I bind with a V8JT module in the X7 the V8R4 LED reacts normally during binding and also after that when switching the receiver and transmitter back on, but there is no signal to the servos on any channel port.

Isaac96, 2017-03-20:
That is the weirdest thing I've ever heard of. The TX code shouldn't change anything with the RX config. Maybe it has something to do with telemetry?

davidppoole, 2017-03-20:
jmpblackfoot,
I don't know if this helps, but:
I'm using the BG 4 in 1 module (purchased March 2017) flashed with V1.1.6.17 firmware,
in a Taranis (original model) with V2.2 RC12 firmware.
I have successfully bound an Frsky V8R4 to it and plugged in one servo to check operation.
I did not range test it or use it in a model.
I calibrated the frequency tuning (set to middle of highest and lowest working setting).
I also did the same with a V8R7 receiver and it worked ok too.

jmpblackfoot, 2017-03-21:
David,
Thanks for the time you spent testing with your receivers.
I tried a V8FR-II and it works both in V8 and D8 mode.
A D6FR works in D8 mode.
And the V8R4 now works, but the V8FR still behaves the same (no servo output).
So, I conclude that it is not a transmitter or 4 in 1 module problem.
The V8R4 must suffer from a cracked solder joint somewhere, while the V8FR must just have given up the ghost for old age ;<)

pascallanger, 2017-03-22:
I'm closing the issue then.
Pascal

closed 2017-03-22 by pascallanger