pascallanger/DIY-Multiprotocol-TX-Module #61: Sanwa SD-10G and FRSky xsr 2016 binding issue

stefanofalconetti, 2017-03-25:
Hi Multiprotocol Team,

I'have been trying to bind my Sanwa SD-10G with an FRSKY XSR 2016 receiver with NON EU firmware for several days now, but with no luck. I have a multiprotocol board from Bangood ([http://www.banggood.com/2_4G-CC2500-A7105-Flysky-Frsky-Devo-DSM2-Multiprotocol-TX-Module-With-Antenna-p-1048377.html]) that I have updated with the latest firmware via the Arduino IDE, few days ago. This RX supports the D16 protocol and the multiprotocol board is connected to my tx via PPM trainer port. This tx port is declared to be PPM8 capable. I'm getting that it has 8 channels on that PPM signal. I have set the dial to #3 on the board, as 3 is FRSKY protocol and tried all possible bind procedure but with no luck. The Rx just turns up solid red and green leds, and nothing else happens. I also tried changing EU and non EU software on the rx, but with no luck. I'm reading that to bind FRSKY there may be to tune frequency through values of -40, 0, 40 but I didn't get where in the code I can do this. I 'd like to also to try to set D16 specific protocol for this receiver in the dialer but I didn't get how to do this. I'm attaching my _Config.h file. Changes I made are commented with "//Sanwa SD-10G"
I have also changed channels order to EATR (#define EATR). I have also reduced max channels to 8 (My TX has 10). What I see when I try to bind is this:

1. I power the Tx on
2. I power on the multiprotocol module pressing the bind button (I added a powet switch). Green led is solid on
3. I keep the rx bind button pressed while powering the rx
4. I release the bind button on the multiprotocol board
5. Multiprotocol board red led blinks very fast 
6. I release the bind button on the rx
7. After some seconds red led on multiprotocol board turns solid red 
 
Unfortunately nothing happen on the rx, like the tx module is not transmitting any signal, or a signal incompatible with the rx.

Do you have any advice for me on how I could fix my _Config.h file to make further tests to bind successfully the FRSky receiver ?

[image]

[_Config.h.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/870294/_Config.h.txt)

Thanks for your help,

Stefano Falconetti

pascallanger, 2017-03-28:
Hi, It looks like it's working for you right? (so it's not an issue)
I think the best place if you want to keep this knowledge somwhere is to submit a pull request in the section "[How to for popular models](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Models.md)".
Pascal

closed 2017-03-28 by pascallanger

stefanofalconetti, 2017-03-29:
That's correct,

if you can explain to me how I cab properly document it, I can copy-paste
in a more suitable place.

Thanks,

Stefano Falconetti.