pascallanger/DIY-Multiprotocol-TX-Module #63: Telemetry not working

wesk1, 2017-03-27:
Hello, i have 4in1 running latest firmware on a 9xr pro, running latest 221 c4 test version from mike, connecting to a micro scisky with clean flight, dsm2 protocol, not sure if board supports telelmetry but i can't get it to work, is it possible to test telemetry on a 4in1 by connecting a usb to serial?

pascallanger, 2017-03-28:
I don't think the dsm2 version of the sciky is transmitting any telemetry.
To check if the telemetry is working fine between the module and your 9XR Pro, you should in the _config.h file:
 - enable TELEMETRY
 - enable MULTI_STATUS
 - disable INVERT_TELEMETRY
In ersky9x, go in the telemetry menu and check invert com1
Now if you've configured everything correctly, you'll see in the protocol menu the module version, the indication by a blinking N/A protocols that are not in the module and the bind button will blink when you launch a bind.
Pascal

wesk1, 2017-03-28:
i checked inver com1 and now i see the version in the bind section and in debug menu i see the tx numbers going up so the module telemetry is working. Thanks

wesk1, 2017-03-28:
Yea i dont think the micro scisky supports telemetry i will try again later.

closed 2017-04-10 by pascallanger