pascallanger/DIY-Multiprotocol-TX-Module #71: Issue when compiling the MultiOrange

LapinFou, 2017-04-09:
Hi,

When using the command `Build_orangetx.cmd`, I got an error message with the latest version.
This issue is coming from the lines 36-39 in Multiprotocol.ino

```C
//Personal config file
#if defined USE_MY_CONFIG || __has_include("_MyConfig.h")
	#include "_MyConfig.h"
#endif
```
After deleting those lines, everything worked.

LapinFou, 2017-04-09:
Here is the error message:
```
In file included from MultiOrange.cpp:142:0:
Multiprotocol.ino:38:43: error: missing binary operator before token "("
 #if defined USE_MY_CONFIG || __has_include("_MyConfig.h")
```

pascallanger, 2017-04-10:
I've implemented a change.
Please confirm it works.

closed 2017-04-10 by pascallanger

LapinFou, 2017-04-10:
I confirm. This did solved the issue.
Thanks! 👍 
```
-------- begin --------

Cleaning project:
rm -f MultiOrange.hex
rm -f MultiOrange.eep
rm -f MultiOrange.cof
rm -f MultiOrange.elf
rm -f MultiOrange.map
rm -f MultiOrange.sym
rm -f MultiOrange.lss
rm -f MultiOrange.o Wmath.o
rm -f
rm -f
rm -f
rm -f
rm -rf .dep
-------- end --------

-------- begin --------
avr-gcc (GCC) 4.8.0 20130306 (experimental)
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling C++: MultiOrange.cpp
avr-gcc -c -mmcu=atxmega32d4 -I. -x c++ -gdwarf-2 -DF_CPU=32000000UL -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -fno-exceptions -Wall -Wa,-adlns=./MultiOrange.lst  -MMD -MP -MF .dep/MultiOrange.o.d MultiOrange.cpp -o MultiOrange.o

Compiling C++: Wmath.cpp
avr-gcc -c -mmcu=atxmega32d4 -I. -x c++ -gdwarf-2 -DF_CPU=32000000UL -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -fno-exceptions -Wall -Wa,-adlns=./Wmath.lst  -MMD -MP -MF .dep/Wmath.o.d Wmath.cpp -o Wmath.o

Linking: MultiOrange.elf
avr-gcc -mmcu=atxmega32d4 -I. -gdwarf-2 -DF_CPU=32000000UL -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Wall -Wno-main -Wstrict-prototypes -Wa,-adlns=MultiOrange.o  -std=gnu99 -Wundef -MMD -MP -MF .dep/MultiOrange.elf.d MultiOrange.o Wmath.o --output MultiOrange.elf -Wl,-Map=MultiOrange.map,--cref     -lm -N

Creating load file for Flash: MultiOrange.hex
avr-objcopy -O ihex -R .eeprom MultiOrange.elf MultiOrange.hex

Creating load file for EEPROM: MultiOrange.eep
avr-objcopy -j .eeprom --set-section-flags=.eeprom="alloc,load" \
        --change-section-lma .eeprom=0 --no-change-warnings -O ihex MultiOrange.elf MultiOrange.eep || exit 0

Creating Extended Listing: MultiOrange.lss
avr-objdump -h -S MultiOrange.elf > MultiOrange.lss

Creating Symbol Table: MultiOrange.sym
avr-nm -n MultiOrange.elf > MultiOrange.sym
avr-objcopy -O binary MultiOrange.elf MultiOrange.bin

Size after:
AVR Memory Usage
----------------
Device: atxmega32d4

Program:   16234 bytes (44.0% Full)
(.text + .data + .bootloader)

Data:        622 bytes (15.2% Full)
(.data + .bss + .noinit)
```