pascallanger/DIY-Multiprotocol-TX-Module #72: PB4 -> PE1 not wroking on STM32 Boards

dschiedsch, 2017-04-09:
Hello
I had a problem that when using my STM32 multi protocol board with the 4 in 1 module 
all the protocols using the CC2500 chip didn't work.
I traced the problem to the PE1 pin not switching (so the RF switch was selecting the wrong chip and the signal wasn't output)
After a lot of search I found that PE1 is connected to PB4 on the STM32. This pin is used for the SWD debug port and is not usable as GPIO in default configuration
I had to add the following line of code:
afio_cfg_debug_ports(AFIO_DEBUG_NONE);
at the beginning of the void setup() routine.
Now the pin toggles as expected and the protocols using CC2500 work as expected.

Using Arduino 1.6.12 with most recent Multiprotocol github version and the most recent version of the stm32 library also from github

dschiedsch, 2017-04-09:
Add:
Issue is documented here from library side:
http://www.stm32duino.com/viewtopic.php?f=35&t=1130

pascallanger, 2017-04-10:
I've implemented the change as you mentionned.
Please confirm it works.

closed 2017-04-10 by pascallanger

dschiedsch, 2017-04-14:
Can confirm it works on my STM32 board (rev. 0.8)
Compiled with Arduino 1.8.2 and most recent STM32 library