pascallanger/DIY-Multiprotocol-TX-Module #85: NRF Set Bitrate resets to low power

soligen2010, 2017-05-07:
The power setting is getting reset when the bit rate is set, but the prev_power variable is not updated accordingly.  This creates a situation where the power can get stuck in low mode.  Just need to set prev_power accordingly
`void NRF24L01_SetBitrate(uint8_t bitrate)
{
    // Note that bitrate 250kbps (and bit RF_DR_LOW) is valid only
    // for nRF24L01+. There is no way to programmatically tell it from
    // older version, nRF24L01, but the older is practically phased out
    // by Nordic, so we assume that we deal with modern version.

    // Bit 0 goes to RF_DR_HIGH, bit 1 - to RF_DR_LOW
    rf_setup = (rf_setup & 0xD7) | ((bitrate & 0x02) << 4) | ((bitrate & 0x01) << 3);
    NRF24L01_WriteReg(NRF24L01_06_RF_SETUP, rf_setup);
    prev_power = NRF_POWER_0;          // Power setting was just reset.  This will get updated in the next call to SetPower
}`

pascallanger, 2017-11-20:
This is not exact...
The first part of the line : "rf_setup = (rf_setup & 0xD7) ..." is preserving the power settings which are stored in the bits 1 and 2 (0x06).

closed 2017-11-20 by pascallanger

soligen2010, 2017-11-21:
Unfortunately this was so long ago that I don't recall the exact scenario that caused an issue, but there was a problem.  Perhaps it was because prev_power is a shared variable used by other chips as well.  Another protocol could change the value of prev_power but rf-setup would not be correct of the protocol was using another chip.

I suggest at least in NRF24L01_Initialize that pre-_power get reset (prev_power=0xFD; // unused power value) becasue rf_setup is set to a contstant 0x09 and prev_power is not synced to this.

In fact now that I think about it, I recall the issue had to do with changing protocols without there being a power cycle, and re-setting prev_power in NRF24L01_Initialize seems like the much more sensible fix since rf_setup is being set to a constant.

You may want to be sure pre-power gets reset when any protocol is initialized since it may have a value for a different chip from the previous protocol.

soligen2010, 2017-11-21:
I don't see how to re-open this issue, but I think it should be as there was a problem.  I think the fix in my last comment is a much better one.  Can you re-open this or should I start a new issue?

soligen2010, 2017-11-23:
I see that prev_power is getting reset elsewhere in the code when the protocol changes.  I know I had a problem at some point with the power level getting out of sync, but cant reproduce the issue.

In the past I was using two different bit rates.  I have since abandoned this approach.  All I can think of is that in the old code I was doing something unexpected when changing bit rate.  Since am not changing bit rates anymore, I cant reproduce the issue.

Sorry for the false issue.

pascallanger, 2017-11-23:
Anyway, if you check the latest code you'll see that I've added 2 lines to prevent rf_setup and prev_power to be out of sync.
Have a look and let me know.
Also I don't really see why you need to redefine the power function in your custom protocol. If that's just for getting power 3 you can change it in the config...

soligen2010, 2017-11-24:
If I change the config, to get power level 3 when power is set to high then I think all NRF24L01 protocols would end up getting level 3 instead of 2.  I assumed you didn't want any of the other protocols to run power level 3 so I didn't want to impact the settings for any other protocol.

I personally have tested and I get better range on level 3, as you have pointed out before, for some people Power level 2 works better.  Perhaps 3 works for me because I am always sure to use a linear regulator and add shielding, whereas others may not. Also as mentioned on the other thread, I am looking to clean up the power even more.

If you want to make High power be level 3, then I can remove my function.  Or did you have something else in mind?

soligen2010, 2017-11-24:
Oh, and those prev_power changes look good to me.  Thanks.