pascallanger/DIY-Multiprotocol-TX-Module #87: Not Binding to AFHDS2A

krocebow, 2017-05-30:
Multiprotocol code version : Can you please tell me where to find it?
TX type : Taranis Q X7
Using serial
Module is blinking when entering binding mode, RX stays in binding mode with a blinking led.

I have a FS-IA6B receiver that don't bind to my transmitter I did the following procedures and none of them  worked :

1. Turn on transmitter, Set RF mode Multi FS 2A, submode PPM/IBUS, AutoBind checked. I activate the bind mode, and then I turn on the RX.
2. I do all the above steps, but this time I turn off the TX, turn on the RX then I turn on the TX so maybe the autobind will work, but it doesn't.

I tried to enter the bind mode first and it didn't work, I also tried to turn on the RX first and it didn't work either.

I already binded two different receivers with my transmitter and they did work : Redcon CM703 7CH PPM using the DMX protocol and the FS-R9B 8CH using ther FlySky STD protocol.

Any help will be appreciated. Thanks!

schwabe, 2017-05-30:
Don't turn on autobind for that protocol

krocebow, 2017-05-30:
I unchecked autobind, I entered the bind mode on my TX and still same result both Leds are blinking and it didn't bind.

krocebow, 2017-06-01:
I managed to get it to work, but I have another issue, the RX & TX are binded, but the RF signal keeps giving the rf signal critical alarm, whether they are close or not the same result, when I go to the sensors page I see this : 
TRSS: ~250
A1 : 5.01V
RQLY : 1
RSSI : -60 dB
RNse : -101 dB
RSNR : 41 dB

Is it something related to the RX or the multiprotocol module?

carlosveloso, 2017-08-31:
@krocebow did you manage to fix that signal issue? I have the same problem with the FS-IA6B

TheTomCutler, 2017-09-08:
Same issue with FS-X6B and lemon satellite

krocebow, 2017-09-09:
@carlosveloso No I still didn't manage to fix it sorry :)

juliogdiana, 2017-10-02:
@krocebow any update on low signal alarm? 
I have same problem here.

KnuckleUpFPV, 2017-10-27:
Same issue

KnuckleUpFPV, 2017-10-28:
Fixed. Switch it from pwm/ibus to ppm/ibus. No more issues.

juliogdiana, 2017-10-29:
But I need pwm.
I do use it on planes with servos

KnuckleUpFPV, 2017-10-29:
Should still be able to talk to your receiver and it should still do pwm. I tried it on my 3d printed fixed wing and its still doing pwm just fine. And no annoying lost recovered alarm.

closed 2017-12-07 by pascallanger

CNCSeb, 2018-03-05:
Hej,
I am sorry to open up that issue again, but the switch to ppm/ibus did not solve that problem for me. Is there something else I can try to get rid of the warnings?
Kind regards,
Sebastian

TheTomCutler, 2018-03-05:
I'll second this, I still have a wailing radio. I've worked round it by making one of my rotary switches the volume control but this isn't ideal. There's no loss in transmission range so I can only assume it's the telemetry signal coming back that's causing the issue.

pascallanger, 2018-03-05:
This is a known issue with OpenTX, nothing to do with Multi.
Here is the link to it: https://github.com/opentx/opentx/issues/5748

pascallanger, 2018-03-05:
As a workaround until this is resolved you can:
 - set the threshold for RSSI like someone is mentionning in the link above but don't ask me how
 - disable specifically the AFHDS2A telemetry

TheTomCutler, 2018-03-05:
I'll look into disabling the telemetry, I wasn't sure where to start. Many thanks for that.

KnuckleUpFPV, 2018-03-05:
This telemetry warning wasnt an issue until they released a version with failsafe. Once I updated the warnings started.

pascallanger, 2018-03-05:
Commenting this line in_config.h should be enough: #define AFHDS2A_FW_TELEMETRY
But you can always comment this one as well: #define AFHDS2A_HUB_TELEMETRY

TheTomCutler, 2018-03-05:
Top work pascallanger, thanks again!

pascallanger, 2018-03-05:
"until they released a version with failsafe. Once I updated the warnings started"
The failsafe coffee that I've released is completly unrelated to the warning you are seeing...

KnuckleUpFPV, 2018-03-05:
I dont think it was your version Im running Pascal. Someone had the code up on rcgroups. I just added it in. Unless you used that sane code. I'm using the firmware from github with the edited code from the rcgroup post. Telemetry was fine no warnings prior. But had no failsafe. Added the failsafe code and now telemetry warnings. Not all the time though. Depends on receiver I'm using. Pwm ibus in tx setting still has no warnings on two of my ia6b. Yet my others alarm like crazy and didn't before hand. Maybe I screwed up adding the code. Not sure. If i find some time I will try the newest version from github. I'm on the version that I cannot use multi telemetry

pascallanger, 2018-03-06:
It means that you are using a quite old code... You should upgrade.
Also failsafe can be disabled from the config file. So it would be good to do a test. I'm still thinking they are unrelated but it does not harm to test it.

pascallanger, 2019-05-08:
This is expected, you need to change the detection threshold on your radio. Ask on rcgroups, people will tell you what they use depending on your TX.
Pascal

michalelektryk, 2019-05-08:
Well i’m using multiprotocol tx

pascallanger, 2019-05-08:
Well I don't know what a multiprotocol TX is. I'm taking about your radio and firmware running on it ersky9x,opentx,...

michalelektryk, 2019-05-08:
DIY Multiprotocol TX and taranix x9d+ with 2.2.3 firmware

pascallanger, 2019-05-08:
As said earlier please post on rcgroups to get the correct settings for warning and critical: https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page925 .

AndreBL, 2019-05-09:
Suggested RSSI alarm threshold settings for AFHDS 2A in OpenTx (Telemetry tab):
Low: 15
Critical: 12

If you find trouble setting these values as Companion does not allow them, try setting them to 20 which is the lowest value allowed by Companion.

Heri201, 2019-07-06:
Hi..

Can ppm flysky AFHDS 2A use multiprotocol this? thank you..

pascallanger, 2019-07-06:
yes

Heri201, 2019-07-10:
Now I'm happy. I can bind to flysky, frsky and other protocols. thank you @pascallanger for this amazing technique.