pascallanger/DIY-Multiprotocol-TX-Module #94: AFHDS 2A Failsafe

AndreasSeco, 2017-08-01:
Hello, I would like to ask about the failsafe for AFHDS 2A. I already had a look in to the AFHDS2A_a7105.ino code (not to deep) but I have seen that the protocol looks like ready for failsafe packet. But in the taranis X9D+ is no option to set the failsafe settings.

Used HW: 
RX: FS-IA6B
TX: Taranis X9D+ (openTX 220)
TX module: IrangeX (STM32)

schwabe, 2017-08-01:
@AndreasSeco I have code in my sync_multi branch that allows settings failsafe from OpenTX side but the AFHDS2A code and multicode in general needs also updated. The branch https://github.com/schwabe/DIY-Multiprotocol-TX-Module/commits/multi_telemetry_v2 contains already part of the parser required for the new failsafe message. 

But someone needs to sit down and actually implement this on the DIY Multi side. If you willing to do that, I can help.

AndreasSeco, 2017-08-02:
Well, I could have a look. Maybe I will be able to help. I have experiences with AVR and STM32. But not at the professional level.

KnuckleUpFPV, 2017-10-28:
Is this available yet. Have my module but will not use it without a failsafe. I had all my receivers programmed from my fs-i6 to failsafe. With my plain flysky module the failsafe settings stay stored in the receiver. However once bound to the multi module it wipes those settings.

RadianceNL, 2017-10-29:
I've found some working code in this topic on rcgroups.
https://www.rcgroups.com/forums/showpost.php?p=37890760&postcount=9517.

Maybe this code could be added to the original code on GitHub?

schwabe, 2017-10-29:
That code is more a hack and proof of concept but yes an implementation could be based on that.

fernandop009, 2017-10-29:
check this page. https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page677

KnuckleUpFPV, 2017-10-29:
That code worka really well. Delay is about 1.5 seconds after failsafe and motors stop. I prefer shorter, but not going to complain. Its better than a fly away.

pascallanger, 2017-12-07:
The latest version of Multi supports failsafe on various protocols including AFHDS2A. By default it's only acting on the throttle channel by forcing it low and all other channels centered. But using Serial and the upcoming (not released to public yet) OpenTX/er9x/ersky9x versions you can set all channels as you want on the radio by model.
Feel free to test at least the throttle cut version for now.

closed 2017-12-11 by pascallanger

Sekilsgs2, 2021-06-30:
Failsafe not working on latest 77 FW and on T12 Pro with 2.3.13 FW - what option i'm need choos to proper working FS with afhd2a???

pascallanger, 2021-06-30:
That would be more than surprising... Have you used the failsafe menu to set the servo positions?

Sekilsgs2, 2021-06-30:
P.S. - I'm test FS when im poweroff my radio T12..

I'm try using many variants - 
1. No pulses
2. reciever
3. custom

When using custom FS - i'm try change FS value for ch3 - but in this menu i'm can only change from -100 to 100 - this not FS values - inav cant see FS...
Where i'm wrong?

Sekilsgs2, 2021-06-30:
> -100 to 100 is a failsafe value. It means that the RX will set CH3 to that position when he is losing connectivity.

And how inav detect FS with -100 (1000) if this normal value for each ch? need values about -120 (~970) or something similar.

pascallanger, 2021-06-30:
Then enable extended limits to be able to go lower... That has nothing to do with Multi or the protocol but learning how OpenTX works...

Sekilsgs2, 2021-06-30:
> Then enable extended limits to be able to go lower... That has nothing to do with Multi or the protocol but learning how OpenTX works...

Very thanks. 
I am very sorry for taking your time.

HazzaHFPV, 2021-06-30:
> And how inav detect FS with -100 (1000) if this normal value for each ch? need values about -120 (~970) or something similar.

I think for INAV to detect failsafe properly having the receiver send no pulses would be the easiert way. You can also define the activation of the failsafe mode in the Modes-Tab of INAV and assign an AUX Channel to it. Example:

Set failsafe in the receiver so that for example AUX4 (Channel 8) goes to +100.
Then define the failsafe Mode in the Modes-Tab of INAV to AUX4 and a range of 1900-2100.

As soon as AUX4 goes to that range the failsafe mode is triggered.
Otherwise INAV will detect failsafe in the following conditions:

- Any flight channel (pitch, roll, throttle or yaw) sends no pulses
- Any channel is outside the valid range between rx_min_usec and rx_max_usec
- The FAILSAFE aux mode is activated