pascallanger/DIY-Multiprotocol-TX-Module #96: Make place on PCB for 9XR internal antenna connector

steelman, 2017-08-07:
Redesign PCBs so it is possible to connect module to 9XR's internal antenna. I can see two possible options.

# Leave a hole for a connector on U.FL pigtail
# Reroute RF and mount the connector on a PCB

Might be the latter prepared so, that on one side there is U.FL for SMA connector and SMB(?) for the internal antenna?

KC5CQW, 2017-08-19:
Replace the U.FL with an MCX (or MMCX not sure which) to match the internal antenna jack. Jumpers are relatively easy to find online and offer much better tolerance to abuse over the U.FL's. The internal jack is outside of the PCB footprint and wouldn't necessarily fit other TX's. Pigtails offer the most flexibility (pun intended).

pascallanger, 2017-11-20:
You should use the 3D model designed for it. It works without an issue.

closed 2017-11-20 by pascallanger

steelman, 2017-11-20:
What 3D model? Of a cover? This requires an external antenna. 9XR has a built in one in the handle with a connector available in the module bay.

pascallanger, 2017-11-20:
I'm using the 9XR Pro antena in the handle with both the STM32 and Atmega versions.
I'm reffereing to this 3D model/print (scroll a little on the page):https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Module_BG_4-in-1.md#what-you-need which is available on thingiverse: http://www.thingiverse.com/thing:2050717 . It works very well.
- Pascal

steelman, 2017-11-20:
I wouldn't have bothered you, if these links were include included above :-) Thank you very much.

pascallanger, 2017-11-20:
Sorry but this in the documentation...