pascallanger/DIY-Multiprotocol-TX-Module #99: Using the DIY V1.2 in the EU

dklueh79, 2017-08-20:
Do you have some infos about the maximum performance in (watt) the diy transmitter module has? It's important for using it in the EU. 

https://www.banggood.com/2_4G-CC2500...p-1048377.html

schwabe, 2017-08-20:
Your link is broken. But the maximum output is controlled by the software. But it should around 100 mW.

dklueh79, 2017-08-21:
Ohh, sorry. This link is right:
https://m.banggood.com/2_4G-CC2500-A7105-Flysky-Frsky-Devo-DSM2-Multiprotocol-TX-Module-With-Antenna-p-1048377.html

Hmm, ok. 100mW would be ok. I'll will search for the option within the scetch.

pascallanger, 2017-11-20:
The power settings for each RF component are set in Multiprotocol.h (middle of the file). They can be changed to whatever you want.

closed 2017-11-20 by pascallanger