pascallanger/DIY-Multiprotocol-TX-Module #102: Help adding new protocols

jacobbloy, 2017-09-16:
Is there any way that support could be added for an electric skateboard? The evolve gt uses nrf24lo1+ and seems to operate the same as an RC transmitter, it even has telemetry.
Because of the cheap way it was made I wouldn't be supprised if It is using an existing protocol.

Would you need both the board and transmitter to reverse engineer or could you do with just the transmitter? The transmitter has a cheap stc15w4k32s4 chip on it which I think I have seen on some china transmitters. Jacob.bloy@gmail.com

jacobbloy, 2017-09-16:
[image]
[image]

pascallanger, 2017-11-20:
We need dumps of what the protocol looks like. You could attach a logic analyzer to the SPI port of the nrf board that I can see on the top left of the 1st picture.
Protocols using the NRF are usually not having telemetry. So I'm thinking it's not a known protocol.

closed 2017-12-07 by pascallanger