pascallanger/DIY-Multiprotocol-TX-Module #104: Hitec AFHSS

Ze-baloo, 2017-10-07:
Hi all!

I've installed the multiprotocol module in my Taranis, and i love all its features.

I'm wondering if it'll be possibe to add th Hitec AFHSS protocol as Frsky's producing compatibe receivers for this brand.

Thank in advance for any reply.

Laurent

schwabe, 2017-11-02:
Yes that would be possible if someone reverse engineers the protocol

Ze-baloo, 2017-11-02:
Thanks for the reply 😉

I'd like to do it myself, but I must say that i will have a huge lack of knowledge.

By the way, I'm curious to know the reverse engineering process, to see if I could find the ressources to do it.

Laurent

pascallanger, 2017-11-02:
It should be a quote easy protocol to reverse. But the problem is to have access to a transmitter and receiver...

tipouic, 2017-11-02:
If you want to know how the protocols are added, here is a link that explains how to do for the HUBSAN protocol
[http://www.jimhung.co.uk/?p=1349](url)
I wanted to do it for L939 WLTOYS car but the problem is the chip (a KN) and the lack of equipment.

Ze-baloo, 2017-11-02:
In fact, I'm selling my Horus to one of my best friends who also owns the Aurora 9 and some hitec receivers. 😉

Ze-baloo, 2017-11-02:
Tipouic, I am not able to open your link...

tipouic, 2017-11-02:
Sorry typing error. Correction made
[http://www.jimhung.co.uk/?p=1349](http://www.jimhung.co.uk/?p=1349)

pascallanger, 2017-11-20:
@Ze-baloo I'm ok to reverse engineer the Hitec protocol but you need to send me a TX and a RX to test on. Feel free to contact me on rcgroups by PM if this is something you are ok with. I'll return the hardware when I'm done with it.

Ze-baloo, 2017-11-20:
@pascallanger Thanks for the proposal. As the radio gear isn't mine, I have to check with my friend. Were are you located?

pascallanger, 2017-11-20:
Near Paris.

Ze-baloo, 2017-11-20:
Fine, I'm in Normandy 60km south from Caen.
Not too far away 😉
How long will you need to do the reverse engineering?

pascallanger, 2017-11-20:
It depends how well it goes... Let's say 2 weeks depending how busy I am. If I'm not done at this point I would only need to keep one RX a little longer to do my tests. I'll be at Courseulles-sur-mer the week of Christmas if that helps.

Ze-baloo, 2017-12-01:
Ooops. Sorry for the delay ;-)
A bit busy with my job and preparing christmas.

My friend is OK with the idea, He just need a bit time to program on the Horus the models already programmed on the Aurora.

By the way, i haven't found the way on github to send you a private message.

pascallanger, 2017-12-01:
You can reach me on RCGroups pseudo hpnuts : https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page704

pascallanger, 2017-12-19:
What's the deal with this Hitec transmitter/receiver?

Ze-baloo, 2017-12-19:
Send me a mail on my private address please 😉l(dot)bassaler(at)free(dot)fr

closed 2017-12-19 by pascallanger

TheRealMoeder, 2018-05-09:
Pascal, any progress on this issue?

pascallanger, 2018-05-09:
Nobody ever sent me a TX and RX to do the reverse engineering....

TheRealMoeder, 2018-05-10:
OK, thanks for the update!