pascallanger/DIY-Multiprotocol-TX-Module #105: [FeatureRequest] Wireless trainer mode

Frans-Willem, 2017-10-23:
Seeing as the multi-module is full of radio modules, and a custom MCU, would it be possible to also implement receiving for one or more protocols ?
This would be ideal for trainer mode. I'd be able to fly my airplane using the built-in module, set the multi-module to receive trainer data, and use any cheap DSM2 transmitter as a slave.

schwabe, 2017-10-23:
Yes, would be possible but if you plan to use that in a Frsky radio you want to have input on the heartbeat/trainer pin, which is not connected on the multimodule.

pascallanger, 2017-11-20:
While it's easy to achieve this is not the goal of this project. You'll find on rcgroups multiple projects doing exactly what you are looking for.

closed 2017-11-20 by pascallanger