pascallanger/DIY-Multiprotocol-TX-Module #109: Going to autobind when receiver isn't switched on

DonKracho, 2017-11-11:
Hi,

I'm using the banggood STM32 4-IN-1 module (flashed to 1.1.6.21 INV) in a Traranis X9D+ (OpenTX 2.2) I have got a lot of ASSAN receivers, so I'm quite happy that the protocol is supported. I can bind receivers and the link is stable.

No the issue: Usualy you have to spwitch on the radio first an then the model. The Autobind radio button in the model page ist switched off, but the Taranis does not care and goes to bind mode. An already bound receiver will not establish the link then. I have to go to the model settings, select another protocol and going back to ASSAN again to stop the bind procedure and the the receiver immediately goes green, saying the link is fine.

How do I prevent the autobind procedure? I have the same issue with DSMX and Devo. When set to a fixed ID the transmitter should not go to bind mode again.

B.T.W is there any way to get rid of this really loud bind beeping?

schwabe, 2017-11-12:
The loud beeping is OpenTX "I am binding" sound. Please open a seperate issue on that. For some protocols (especially toy protocols) the protocols need to bind at start to work, so on these protocols autobind is unconditionally enabled. However for DSMX, Devo and Assan this is not the case. Are you using serial?

DonKracho, 2017-11-12:
Yes, I'm using serial. I compiled the source with Arduino/ST32 toolchain because I wanted TAER channel order. I also tried to use autobind in conjunction with auto bind channel 16 feature but even with no low-high transition on chanel 16 the bind procedure starts. It happens not all the time but in about 80% of the power cyples of the Tarranis radio. So it looks like a timing issue or race condition. May be it is ralated to the Taranis firmware. 

Settings of _config.h
#define TAER
#define ENABLE_BIND_CH
#define BIND_CH	16
#define WAIT_FOR_BIND

pascallanger, 2017-11-20:
What's the Autobind setting for this protocol for the specific model?
What's the LED status of the module when you startup the TX?

DonKracho, 2017-11-28:
Looking for the module LEDs has been a great idea! The problem is solved! It has been a mechanical issue. Removing the cover of the module tray did stop the unwanted 'auto bind' feature  immediately.  The cover did press the bind button slightly. Sometimes it did make a contact, sometimes not. My fault. This issue can be closed. Thank you. This module is a fun and really great work analyzing and implementing all the protocols!

pascallanger, 2017-11-28:
Feel free to donate if you are enjoying the multi module to support it's future development.

closed 2017-11-28 by pascallanger