pascallanger/DIY-Multiprotocol-TX-Module #125: Blade 350 Qx2 with iRangeX irx4 and Taranis Qx7

theholzixx, 2017-12-14:
When trying to use the irx4 with my Blade 350 Qx2 i cannot get it to arm, even though it's working fine with the original remote. When the reciever is connected to betafligth through an F4 V5PRO the channels go only from 1090 to 1908 which is probaply why the qx won't arm given that it thinks that the throttel isn't all the way down.
If this is a known issue with spektrum satellites then i'm sorry to have brought it up again but would be thankfull to be redirected to the solution, as i was unable to find anything myself.

pascallanger, 2017-12-15:
1100 to 1900 is the 100% range of DSM which should correspond to the 100% range of your taranis.
If you want more range, you need to enable 125% on your taranis and change the limits of the channels to 125%.
Pascal

theholzixx, 2017-12-15:
Thank you Pascal for the fast answer but after looking at what the original Remote is sending i found that it only goes from 1160 to 1840 wich corresponds to 83.3% of my taranis and only puts channel 6 to 988 if the disarm button is pressed. After configuring the Taranis to match this it worked fine. I quess the 350qx just wants to be special or something but given that the Reciever works fine, this will only be a problem for 350qx users.

closed 2017-12-15 by theholzixx