pascallanger/DIY-Multiprotocol-TX-Module #129: can't compile

jeanbat2, 2017-12-17:
Hello, I can't finish the compiling on the arduino IDE : 
There is a lot of "not declared" issue, for example :

Telemetry:174: error: 'Serial_write' was not declared in this scope

D:\documents\0modÃ©lisme\multiprotocole\DIY-Multiprotocol-TX-Module-master\DIY-Multiprotocol-TX-Module-master\Multiprotocol\Telemetry.ino: In function 'void frskySendStuffed()':

Telemetry:192: error: 'Serial_write' was not declared in this scope

D:\documents\0modÃ©lisme\multiprotocole\DIY-Multiprotocol-TX-Module-master\DIY-Multiprotocol-TX-Module-master\Multiprotocol\Telemetry.ino: In function 'void sportSend(uint8_t*)':

Telemetry:503: error: 'Serial_write' was not declared in this scope

D:\documents\0modÃ©lisme\multiprotocole\DIY-Multiprotocol-TX-Module-master\DIY-Multiprotocol-TX-Module-master\Multiprotocol\Telemetry.ino: In function 'void sportIdle()':

Telemetry:683: error: 'Serial_write' was not declared in this scope

exit status 1
'random_init' was not declared in this scope

And here is the failed message when I try to Burn the bootloader :

avrdude: Warning: cannot open USB device: Function not implemented
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'
Erreur lors de la gravure de la séquence d'initialisation.

Can you help me please?

pascallanger, 2017-12-17:
Hi Jeanbat,
Describe your setup: which module? which radio?
Explain what you've changed in your_config.h
Tip: remove the accents from the path since they could cause issues "D:\documents\0mod**Ã©l**isme".
PAscal

jeanbat2, 2017-12-17:
Hi, my module is the 4 in 1 with atmega 328, my radio is a T9x with er9x.
I set up the table with the protocols I want.

Since my first post : I disabled the protocols I don't want on the define section by commenting.
I disabled the telemetry : telemetry issues are gone, but I still have this issue : 

D:\documents\0modÃ©lisme\multiprotocole\DIY-Multiprotocol-TX-Module-master\DIY-Multiprotocol-TX-Module-master\Multiprotocol\NRF24l01_SPI.ino: In function 'void NRF24L01_Activate(uint8_t)':

NRF24l01_SPI:123: error: 'SPI_Write' was not declared in this scope

exit status 1
'random_init' was not declared in this scope

I will try to delete the accent

jeanbat2, 2017-12-17:
Ok so the accent resolved the compile problem, now I have the bootloader issue, can I upload it to the board anyway?

pascallanger, 2017-12-17:
The previous message you sent is pretty clear: avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de'
It does not find your usbasp...

pascallanger, 2017-12-17:
Why are you using ppm on er9x which supports serial?

closed 2017-12-18 by pascallanger