pascallanger/DIY-Multiprotocol-TX-Module #130: win / do_version.bat errors

mozgy, 2017-12-17:
"\AppData\Local\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.2/tools/win/do_version.bat" "\AppData\Local\Temp\arduino_build_22481" "Multiprotocol.ino" "\Documents\Arduino\Projects\Multiprotocol" MULTI_STM32_NO_BOOT
find: #define VERSION_MAJOR: No such file or directory
find: #define VERSION_MINOR: No such file or directory
find: #define VERSION_REVISION: No such file or directory
find: #define VERSION_PATCH_LEVEL: No such file or directory

benlye, 2017-12-17:
What version of the Arduino IDE are you using? 

It looks like the problem is that the IDE is passing relative paths but the script expects full paths. I'm sure I can fix it, but it would be helpful if I could repro it myself too.

benlye, 2017-12-17:
For example, when I build get a command which looks like this:
`C:\Users\blye\AppData\Local\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.2/tools/win/do_version.bat" "C:\Users\blye\AppData\Local\Temp\arduino_build_277497" "Multiprotocol.ino" "C:\Users\blye\GitHub\DIY-Multiprotocol-TX-Module\Multiprotocol" MULTI_STM32_NO_BOOT`

In `do_version.bat` you could try uncommenting this line : `REM SET DEBUG=1` (remove the `REM`).  That will cause it to give a bit more output which will help me.

benlye, 2017-12-17:
Actually, it seems like you are running a different 'find' command to the one I expect.  I guess there is some tool (maybe Cygwin) or something else putting an alternative find.exe higher up.

What output do you get if you open a command prompt and run `find /?` ?

benlye, 2017-12-17:
Reasonably sure I've got a fix by hard-coding the path.  Also updated to use findstr.exe instead of find.exe.

You could test by replacing your copy of 'do_version.bat' (found in `%localappdata%\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.2\tools\win`) with the content of the attached text file.

[do_version.bat.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/1565976/do_version.bat.txt)

If you can confirm that this fixes it I'll update the board manager packages.

Ben

mozgy, 2017-12-18:
Here they are -

```
C:\>where find.exe
C:\Program Files (x86)\WinAVR\utils\bin\find.exe
C:\Windows\System32\find.exe
```
with fixed version it works and I do get -
```
"\AppData\Local\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.2/tools/win/do_version.bat" "\AppData\Local\Temp\arduino_build_975499" "Multiprotocol.ino" "\Documents\Arduino\Projects\Multiprotocol" MULTI_STM32_NO_BOOT
Sketch Path: \Documents\Arduino\Projects\Multiprotocol
Multi board: MULTI_STM32_NO_BOOT
Getting Multi-MODULE firmware version from "\AppData\Local\Temp\arduino_build_975499\sketch\Multiprotocol.h"
Multi-MODULE firmware version: 1.1.6.44
COPY "\AppData\Local\Temp\arduino_build_975499\Multiprotocol.ino.bin" "\AppData\Local\Temp\arduino_build_975499\multi-stm-1.1.6.44.bin" /Y
```

ty, closing,

closed 2017-12-18 by mozgy