pascallanger/DIY-Multiprotocol-TX-Module #131: Custom Fail Safe not saved in OPENTX 2.2.1 on FRSKY X10

fabioeh, 2017-12-18:
Hi Pascal 

The system I am using is 

Radio: FRSKY X10 V2 OPENTX 2.2.1

Irangerx IRX4 stm34 1.1.16.44 

Receivers : FlySky FS-IA6 FLYSKY AFHDS2A - 28 

When I Set the Custom Fail Safe to small left turn and some up elevator the setting is getting lost.

The rudder and the elevator are set to center.

Please fix the 1.1.16.44  version to respect the failsafe custom setting in OPENTX to be compatible with FLYSKY FS-I6 transmitter failsafe capabilities.

Thanks

pascallanger, 2017-12-18:
What do you mean by the setting is getting lost?

Does it work if you set failsafe on the radio with the RX powered on and wait 10-20sec and then power off the transmitter?

- Pascal

fabioeh, 2017-12-18:
That worked. 

Waiting 30sec and then turning off the Treansmitter .

Then the rudder and elevator worked  as expected.

Thanks for the quick response.

closed 2017-12-18 by fabioeh