pascallanger/DIY-Multiprotocol-TX-Module #135: Additional protocols in PPM mode?

pinkywafer, 2017-12-29:
The iRange IRX4 PLUS firmware as supplied includes the functionality for 30 protocols and serial to be selected using the protocol knob.
The protocols are split into 2 sets - Set 1 (green light off) which has serial mode in position 0, and set 2 (green light on)
The protocol set is switched by long pressing the bind button at module power on.
Is it possible to add this functionality?  or if it is already included, how do I configure the second set of protocols? - only 15 are definable in _Config.h

Thanks

pascallanger, 2018-01-02:
As iRange have added it on their own, it would have been great that they at least share their improvements... This is unacceptable in my eyes... They are making money of our work, do not contact us to propose anything not even a sample, modify the source files and do not share them back which is breaking the open source licence!!!

This feature does not currently exist and I'm not really keen at this moment to add it...

Pascal

pinkywafer, 2018-01-02:
I understand entirely your feelings,  and agree with you.

pascallanger, 2018-01-02:
Where is this green LED connected on the STM32? It's not on the original design...

pinkywafer, 2018-01-02:
I am not sure as I am waiting for delivery of the module. I expect that I will have to reflash anyway in order to be able to have the configuration I want, so I'll have a look when I've got it open for soldering header pins

mlmflip, 2018-01-05:
On the iRange IRX4 PLUS module, can you at least reconfigure protocol assignments for the first 15 switch positions in _Config.h and then compile/upload updated firmware?  Or does a firmware re-compilation wipe out or corrupt the useability of the the second set of protocols when the green light is on (equivalent to switch positions 16-31)? 

Bottom line issue for me:  I have the iRange IRX4 PLUS module and am using PPM.  I need to use the H20 protocol, but it isn't available by default on one of the 30 dial positions out of the box.  I want to reconfigure the dial assignments to allow use of H20.  Is this possible?

pascallanger, 2018-01-05:
You have to ask iRange to make the modification for you...
If you update the firmware then you will loose 16-31.

schwabe, 2018-01-05:
@mlmflip You should demand the modified Firmware from IrangeX, the GPL states they MUST provide it to you.

mlmflip, 2018-01-05:
Understood.  Thank you for your comments, gentlemen.

pinkywafer, 2018-01-05:
I have twice asked for both the compiled firmware and the source code by sending a message using the form on iRangeX's website. I have received no reply to either message

mlmflip, 2018-01-05:
Thanks, Pascal.  As use of this module becomes more widespread, hopefully iRangeX will eventually provide the compiled firmware to allow modification will full use of positions 1-31.  I actually thought this module version with the additional dial options would be an upgrade from the Banggood 4-in-1, but at this point it's actually a downgrade in terms of flexibility for the end user.

pascallanger, 2018-01-05:
@mlmflip : do you have one of these iRangeX modules?
If yes, on which STM32 pin is connected the green LED?

mlmflip, 2018-01-05:
Yes, I recently received the iRangeX module.  I have successfully installed it in my Jeti transmitter and it is working well (see attached pics).  I am at work now, but will check the pin assignment for the green LED later today and advise.

[image]

[image]

tipouic, 2018-01-11:
Hi Pascal,
I just received an STM32 module with USB and selector.
The green LED is an addition on the circuit (pin n°12, PA-2), it is marked STAUS. 

They use the following procedure to switch from 0-15 to 16-31 (witch the green led is off).
Turn off your transmitter, point the protocol switch to "0", hold and press the bind of RF module then turn on your transmitter, after 3 seconds, relase the button.

pinkywafer, 2018-01-11:
From looking at the board, I am unable to tell where the LED is connected. Tracing from the LED goes to a smt resistor, then the trace goes towards the STM32, then it goes through to the other side of the board. The circuit hole is right in the corner of the STM32 by pin 1.  I cannot follow on the other side as it is hidden beneath the RF and I have not got test equipment available to trace.
For reference, don't know if it's any help, the board is printed with the code YRFM_ARM_V1.2

pascallanger, 2018-01-27:
The LED2 has been added to the latest build (on at startup so nothing fancy).
I'm looking at implementing a bank switch but not the way they've done it as I want it be compatible with all modules (which do not have a second LED anyway).

pascallanger, 2018-01-29:
I've implemented the bank switch system when using PPM. Up to 4 banks of 14 protocols can be configured in _config.h which gives a total of 56 entries (more than enough I think). The number of banks can be set from 1 to 4 in _config.h.

Does someone have an idea of which protocols I should set by default in the banks?

Is someone willing to test the firmware on Atmega? I don't have a module around...
I'll test tomorrow the STM32 version but if someone is willing to test as well...

It works like this:
1. turn the dial to 15
2. power up the module
3. the module displays the current bank by flashing the LED x number of times, x being between 1 and up to 4
4. a short press on the bind button turns the LED on for 1 sec indicating that the system has changed the bank
5. repeat operation 3 and 4 until you have reached the bank you want
6. power off
7. change the dial to the desired position
8. power on
9. enjoy

Pascal

pascallanger, 2018-01-30:
No one is willing to test so closing this issue.

closed 2018-01-30 by pascallanger

pinkywafer, 2018-01-30:
Only just seen the update to this. I'll be happy to test on stm32, but do not have an atmega.