pascallanger/DIY-Multiprotocol-TX-Module #136: Betaflight commands not working

thjubeck, 2017-12-31:
Hey, is there anything besides the SPORT_POLLING and invert telemetry that needs to be enabled to get betaflight lua scripts to work?

schwabe, 2017-12-31:
Depends on your remote. For OpenTX you currently need a nightly. For INVERTED telemetry you might run into the problem that the code currently does not switch off the inverter when sending leading to suboptimal voltage level.

(rest is copy & pasted from my [post on rcgroups](https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page717#post38816707) to have the information here in Github)

[image]
I tested with a Banggood with USB Board and SPORT Polling enabled:

There is another much bigger problem at the moment on my Taranis. The telemetry output of the Taranis does not drive the sport line high enough. The 0,75V peak for the output is not really much. Even the input voltage from the multi module with 2V is not really great but good enough for the inverter on the Taranis to register. I attached a screenshot from my Salae.

Update: The IrangeX Board (which uses a normal 2port inverter) has the same problem just with 1,9V and 0,95V.

With the internal XJT module both input and output are 3,25V. Something on the multi board is pulling the voltage levels down but only if the multi module is also powered.

Inverter circuits inside the X9D+ have a total of 10k pull down that is always active. The multi module has only a 4k7 pull that also needs to go through a diode. Ignoring the diode we get 2,2V, the diode could be another 0,2V. So that fits the voltage levels I see. 

When the Taranis drives the sport line, uart output of multi module is high level so, with TX_INV enabled the inverter pulls down sport and that inverter is capable of at least 24mA current. Without the diode's forward voltage drop the Sport line would probably just stay at 0V.

Just toggeling TX_INV in rx_pause/rx_resume does not work. USART is still sending when the inverter is toggled. Probably need to toggle on USART TX finish interrupt

kimman1, 2018-07-21:
Does it work ? . I try STM32 module with 2-transistor inverter but it isn't work

schwabe, 2018-07-21:
@kimman1 the 2 transistor inverter is also not bidrectional, so it cannot work.

kimman1, 2018-07-22:
@schwabe so what i need to get it work ?. 
I think i need a new inverter diagram ? Or we force to use inverter chip in atm diagram.

schwabe, 2018-07-22:
@kimman1 the broken inversion I am talking about in my post is already with the STM32 board and the inverter chip. But I have no working bidrectional inverter diagram at hand.

kimman1, 2018-07-23:
Yep, i have oder inverter chip at TI. Hope it working.
I don't know why because midelic told me that he use STM32 DIY module with 2 transistor invert with his 9xpro TX and betaflight command from his TX work perfect

pascallanger, 2018-08-06:
Please test v1.2.0.26 which should fix sport polling and report.

kimman1, 2018-08-07:
I've tested it with my X9D+ with OpenTX. It's not work

pascallanger, 2018-08-07:
Are you running the latest ersky9x version R222b6 available here http://openrcforums.com/forum/viewtopic.php?f=7&t=4676 ?

pascallanger, 2018-08-07:
@schwabe Will you implement on OpenTX the same fix that Mike applied on ersky9x to get sport working with multi?

schwabe, 2018-08-07:
@pascallanger To my understanding that would be implementing software serial do overcome the signal level problem that is present with the inverter being active.

It might be worth looking into if it is possible to disable TX_INV while the multimodule is listening for S.Port input. That should fix the problem with the low signal level. As you can see from my image in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/136#issuecomment-354606718 OpenTX already answers with Multi telemetry enabled but the signal is pulled down by the inverter and the diode to the forward voltage of the diode (~0.7V).

MikeBland, 2018-08-07:
@schwabe "if it is possible to disable TX_INV" This is what is now done in the module, for 3mS after starting to send the SPort poll. ersky9x is working on the Taranis using the hardware serial with this in.
The downside is you might get a line break detected you need to make sure you ignore.
In ersky9x, I use the same code as for the XJT module. It detects the SPort poll (very likely after half of the first stop bit) and immediately enables the Tx output (line marking so a low voltage with the inverted signal).
To make sure any line break is cleared and the serial data is sent correctly, I turn the TE bit off, then on in the USART control register. This causes the USART to send a whole character (12 bits) of line marking before the start of the first byte, so clearly terminating any line break.

@kimman1 What script are you using with ersky9x?
I don't have any Betaflight hardware, but I tested configuring a S8R receiver using both a MPM module and the internal XJT.

kimman1, 2018-08-07:
@pascallanger Yep. I use ErskyR222b6 for testing

schwabe, 2018-08-07:
@MikeBland I did not see the fix to Multi code. Yes that is exactly what I had in mind. I need to double check what now happens with OpenTX and why it is not working. But that might take a while (life gets in the way)

pascallanger, 2018-08-07:
@kimman1 What receiver are you using, and, with ersky9x, are you seeing a RSSI value?

filippov-shurik, 2019-03-11:
Good day, I started with the same problem on the jumper t12 equipment with a 4 in 1 multimodule telemetry works without problems, but the scripts do not work, how much do I understand the problem in the module and its levels ??? and is there a solution to the problems.

multimodule version 1.2.1.28
I use the r-xsr transmitter connected in fport and sbus/sport
fc omnibus F4. 
betafligh or inav.

pascallanger, 2019-03-11:
It should work using ersky9x. The OpenTX fix has never been implemented.
Personnaly I don't have the hardware to troubleshoot...

delta575, 2019-09-16:
I'm experiencing a similar issue, lua script for betaflight wont work "script syntax error telemetry protocol not supported"

My setup:
Taranis x9d+ (opentx-x9d+-multimodule-noheli-lua-luac-en-2.2.4)
Jumper jp4in1 (multi-stm-opentx-inv-v1.2.1.76)
FuriousFPV LR1000 Diversity Mini Receiver for Frsky
BETAFLIGHTF3 board

pascallanger, 2019-10-03:
Multi and OpenTX dev versions in action: https://youtu.be/eCnxs3d4VKg
It's not released yet (couple of weeks).

MikeBland, 2019-10-09:
Hi
Pascal,

I've received a communication suggesting the PPM bank selection may not
be working on the current Github sources.
I'll try to test this later today, I'll need to check I have the latest
sources and build/flash the firmware to test, but you may have a module
ready to test.

Also see: https://www.modelflying.co.uk/forums/postings.asp?th=145276&p=1#PostTop

Mike

pascallanger, 2019-10-09:
Hi Mike,
This is strange. I wanted to post on the site but you need to create an account.
I'm wondering if the guy has ut more than one bank in the _config.h file since the default is #define NBR_BANKS 1
Pascal

pascallanger, 2019-10-16:
Fixed in Multi v1.3.0.xx and OpenTX 2.3.2 .

closed 2019-10-16 by pascallanger