pascallanger/DIY-Multiprotocol-TX-Module #140: UM BNF Problems

sonorarc, 2018-01-19:
Is there any way to bind the IRX4 STM32 to my 4 channel ultra micro models from horizon hobby P-51D with AS3X and Parkzone P-51. When bound manually from with in the protocol menu, the bind light goes solid and the rud and elv work but the ail and motor operate erratically and the motor turns off and on randomly and cuts out at high throttle positions. It is set to dsm/dsm2-22 or dsmx-22 on the module and manual bind with auto bind both off and on. Channel order is aetr. All my other spektrum/orange/lemon products bind properly with dsm2-22 and all 4 channels work as they should.  I had the same problems using the orangerx module.  Both models bind properly with my DX6i.  I would love to fly these things with my 9xrPro because of its flexibility. What am I doing wrong?

sonorarc

pascallanger, 2018-01-20:
Have you really used auto-detect and you've seen it updating the model parameters (DSM and number of channels)?
If not and as said in the [documentation](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm---6) you need to manually test all the combinations:

> model/type/number of channels indicated on the RX can be different from what the RX is in fact wanting to see. So don't hesitate to test different combinations until you have something working. Using Auto is the best way to find these settings.

Usually even if small models are 4 channels they expect 6-7 channels.

xjrphilippe, 2018-01-21:
Hello,I am the same problem, with as3x, channels configuration spektrum is taer but smt32 default is aetr. How to change smt32 please?Best regardPhilippe

sonorarc, 2018-01-21:
"..... Using Auto is the best way to find these settings."

When you speak of "Auto" are you referring to the Auto in the sub-protocol menu under the DSM2-22 Protocol?  Would the remaining sequence then be, power up the model and press "Bind" with auto bind off?

sonorarc

pascallanger, 2018-01-21:
@xjrphilippe you need to compile your own code with #define TAER in _config.h.
Really simple to do...

pascallanger, 2018-01-21:
@sonorarc 
Have you tried at least manually all possibilities? number of channels from 4 to 8 using DSM-22/DSM2-11/DSMX-22/DSMX-11?

pascallanger, 2018-01-22:
@sonorarc 
The latest version 1.2.0.13 is fixing DSM "Autodetect" when used along with OpenTX.
Have you tried anything on your side?

closed 2018-01-23 by pascallanger