pascallanger/DIY-Multiprotocol-TX-Module #141: Burning bootloader on macOS not working

C0br4, 2018-01-22:
I tried to upload the usb bootloader to my module with no luck. Double checked Board/Port/Upload method and Programmer, everything as it should be.

`Arduino: 1.8.5 (Mac OS X), TD: 1.40, Board: "Multi 4-in-1 (STM32F103CB), Upload via USB"

Intel HEX ERROR: System Error
Error while burning bootloader.
stm32flash Arduino_STM32_0.9
**No such file or directory**

http://github.com/rogerclarkmelbourne/arduino_stm32
`

benlye, 2018-01-22:
Can you enable verbose output and see what the command line looks like when it fails?  It should be easy to fix if I can work out where the path is breaking.  I need to try to get my hands on a Mac to test with.

C0br4, 2018-01-22:
The verbose output reveals `/Users/C0br4/Library/Arduino15/packages/multi4in1/hardware/STM32F1/1.0.3/tools/macosx/serial_upload cu.usbserial-A9M9DV3R 0x0 /Users/C0br4/Library/Arduino15/packages/multi4in1/hardware/STM32F1/1.0.3/bootloaders/Multi4in1/StmMultiUSB.bin ` which should be ok!

the serial_upload script is expecting the file to be the fourth parameter but it is the third.
Also the serial_upload scripts programs to the hardcoded address of 0x800000 but shouldn't the bootloader be at 0x0 ??

benlye, 2018-01-22:
Yep you're right.  It looks like I missed updating the macOS scripts when I modified the Linux scripts.  I'll get that patched up.

closed 2018-01-29 by pascallanger