pascallanger/DIY-Multiprotocol-TX-Module #143: Determine version

sonorarc, 2018-01-26:
How do I find out what software version is in my module?

pascallanger, 2018-01-26:
If you have ersky9x or OpenTX and the telemetry with status is active then the radio indicates it on the protocol screen. Otherwise there is no way to know.

closed 2018-01-26 by pascallanger

sonorarc, 2018-01-26:
Please forgive my ignorance, I'm a hardware user and not very technical.  I'm using ersky9x r220 in a 9xrpro.  How can i determine if telemetry is active with status. I have a telemetry entry on the model's main menu with telemetry, custom and logging sub-menus.

pascallanger, 2018-01-26:
I can't remember when the multi status has been added to ersky9x... You might want to use the latest version available [here](http://openrcforums.com/forum/viewtopic.php?f=7&t=4676).
Then if you have compiled with these parameters defined TELEMETRY and MULTI_STATUS and INVERT_TELEMETRY, you should see the version in the protocol menu near the bind button. It also adds the possibility to see from the radio when the module is in bind mode (Bind blinks), if failsafe is supported by the loaded protocol (otherwise you have N/A) if the protocol loaded exists in the module,..,

sonorarc, 2018-01-27:
Updated to latest ersky9xr and found version by the "bind" in the protocol menu.  It's 1.1.6.21.  Is this version the current one?

pascallanger, 2018-01-27:
Please use the [rcgroups forum](https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page753) for this kind of inquiries... Latest version is 1.2.0.15.