pascallanger/DIY-Multiprotocol-TX-Module #144: USE_MY_CONFIG does work for STM32?!?

LapinFou, 2018-01-27:
Hi Pascal,

In the __Config.h_, there is this comment:
```
/********************/
/*** LOCAL CONFIG ***/
/********************/
//If you know parameters you want for sure to be enabled or disabled which survives in future, you can use a file named "_MyConfig.h".
//An example is given within the file named "_MyConfig.h.example" which needs to be renamed if you want to use it.
//To enable this config file remove the // from the line below. It's automatically loaded if the file exists for the AVR platform but not STM32...
//#define USE_MY_CONFIG
```
The interesting part is:
**It's automatically loaded if the file exists for the AVR platform but not STM32...**

I think, this is not true. I'm successfully using it with my STM32 multi.
Uncommenting the line will take in account my personal parameters.

So, maybe the comment need to be updated? Or did I missed something?

BR,
Seb

pascallanger, 2018-01-27:
Humm may be it's badly written but:
> //To enable this config file remove the // from the line below.

It means that if you remove the "//" it will enable myconfig on AVR and STM32.
> It's automatically loaded if the file exists for the AVR platform but not STM32...

It means that:
 - on AVR if the file exists, it will take it even if the line still has "//"
 - on STM32 if the file exists, it won't be used unless you remove the "//"

pascallanger, 2018-01-27:
May be I should remove the second portion to remove confusion as well as the auto integration of myconfig for AVR...

LapinFou, 2018-01-28:
As you know, English is not my native language. However, I really didn't understand it the way you explained.
If you replace:
**//To enable this config file remove the // from the line below. It's automatically loaded if the file exists for the AVR platform but not STM32...**
with:
**// On AVR boards if the file exists, "_MyConfig.h" will ALWAYS be use even if the line still has "//"
// On STM32 boards if the file exists, "_MyConfig.h" will be use ONLY if you remove the "//"**
Just my 2cents.

pascallanger, 2018-01-29:
I've removed the auto load of _MyConfig.h for AVR to remove confusion.
It will be in the next version.
Pascal

closed 2018-01-29 by pascallanger

LapinFou, 2018-01-29:
Thanks again for all your work.
👍