pascallanger/DIY-Multiprotocol-TX-Module #146: Add additional support methods

JamesHagerman, 2018-02-05:
I'd love to build one of these from scratch instead of paying (and waiting) for a ripped off clone. That means sourcing the parts. While I could get three boards from OSHPark (and also support them) I'd still need all the stuff on the BoM.

Any chance of putting kits together for this and tossing em on Tindie so those that want to support the project have an easier way to do so?

Thank you, all the developers of this module!

pascallanger, 2018-02-06:
Sadly I won't have time to look into this... As you pointed out, everything is available on this site if you want to build your own. About "waiting", you still need to purchase a 4in1 RF module (STM32) or independant modules (AVR) ...

If you want to support this project you can always [donate](/docs/Donations.md).

Pascal

closed 2018-02-07 by pascallanger

okcub2010, 2018-02-09:
I purchased one of the newest BG IRangeX + 4in1 with the yellow knob, ppm and serial capable. Wanting to change the default ch order to TAER, I flashed the multiprotocol fw posted with the _config.h file modified to TAER. Now my module is unusable. Is it possible IRangeX has modified your code and it’s not compatable with this modules hardware?
No reply to requests for info to IRangeX.
Thank you

pascallanger, 2018-02-09:
No the module has no difference and a lot of people are using this module. You must be doing something wrong...
Explain what you see (module lights) and what you do...

okcub2010, 2018-02-09:
This is the module.
If we agree it’s ok with your code, I would line help with the IDE files I need to use. 
The  config file doesn’t seem to match the assign sw positions listed on the + 4in1 info sheet?
I’m just trying to reload/restore the original fw now.

pascallanger, 2018-02-09:
The protocols affected to the switch positions are in the file _config.h (at the end). If you don't configure them for sure it won't match anything...

okcub2010, 2018-02-09:
If posable, could you please post a config file that’s modified to match the info sheet I sent you?
I’m struggling to understand how to change or incorporate changes. 
Also, with many attempting to add a bootloader, they like myself, may overwrite the supplied fw in this or the serial only  IRangeX module and want to restore the original fw. 
Thank you for not ignoring my request for help so far.

okcub2010, 2018-02-09:
> If you spend 5 minutes looking at it I'm sure you'll find how it's working by yourself.
> If you want to restore the original firmware ask iRangeX.

Thank you I’ll try.
IRangeX does not any emails for 2 weeks and their phone # has too many numbers to be valid anywhere. Banggood thinks I want a refund and misunderstands request for firmware or IRangeX contact that works.
MikeB has been coaching me so maybe he can help.
Thank you

JamesHagerman, 2018-02-11:
@pascallanger Thanks for the reply! I fully understand the time challenges involved with such a task.