pascallanger/DIY-Multiprotocol-TX-Module #147: PPM documentation is currently misleading at different links

xtremet1, 2018-02-11:
Pascal, you had changed considerably, the "nominal" originally bought module behavior
with "Position" 15, counting the blips of the red led, changing the behavior of the green led 

At different pages of documentation, one can understand different things & this is missleading
e.g. https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/PPM_Setup.md

vs https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md

or your description at the forums of : 
It works like this:
1. turn the dial to 15
2. power up the module
3. the module displays the current bank by flashing the LED x number of times, x being between 1 and up to 4
4. a short press on the bind button turns the LED on for 1 sec indicating that the system has changed the bank
5. repeat operation 3 and 4 until you have reached the bank you want
6. power off
7. change the dial to the desired position
8. power on
9. enjoy

Pascal  - https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page756#post39094169

Further, one must have the ability & understanding to read the "Config" file, and understand that OLD directions of for example BANGGOOD product  description should be totally ignored.

It's not just turning the dial anymore, and "No Zero" Position for "Greed LED" to go off, whole logic had changed some what like described in : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md

However - I think that correction need to be done in all pages as currently it's missleading
IF I can help you with that please tell me

pascallanger, 2018-02-11:
Could you rewrite the doc as you think will be easier?
Same for _Config.h if you think it's not simple enough...

> Further, one must have the ability & understanding to read the "Config" file, and understand that OLD directions of for example BANGGOOD product description should be totally ignored. It's not just turning the dial anymore, and "No Zero" Position for "Green LED" to go off, whole logic had changed some what like described in : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md
Only the iRange RX4**+** has this bank switch feature with the green led. All others do not have a second LED and no bank switch...

> Further, Binding behaves differently (Much shorter times... e.g. sync of FrSky ..v911 heli
The code has not changed this behavior at all. I think you are mistaken here...

Pascal

pascallanger, 2018-02-11:
I've updated the different documentations and _config.h .
Let me know if that's enough or propose your changes otherwise.

Pascal

xtremet1, 2018-02-11:
Pascal, I think I have at least 3 comments, however I would not like to
comment when I'm tired as I was debugging
what had happened to the Multi-Module till 2AM last night

My 4 Comments in general are :
1) Indicate about the "Green Led Behavior" (as you had changed that vs
Stock)
2) At Protocols page top - I think a link should be made explaining the a
5x14 Table is created at the Confing File - which this one controls
actually all the detailed below protocols  (later below I would also detail
about DSM2/DSMX 1024/2048 etc... as where those relate to... & give example
like AR6100... or other common receivers as to where they belong to & how
that's connected to the Config)  - I may not explain my self well here (as
I'm very tired :-) )) if I cut from your site put in WORD & make changes in
Track Changes - would that be O.K ?
3) I would note (on many pages) - e.g. HW page and others that when you
"Burn In" your firmware - that's it's !   you need to use "new tables" all
OLD tables are gone & never return  (of if I didn't get it right... maybe
you can indicate the "Base" BANGGOOD is programmed with
4) BINDING - I have small correction + If you can indicate the "Time" you
have when the "BIND Button" is presses & later when it's released (from the
TX / Modules - Prespective

What I had seen ... the v911 was "Easy" "Very Fast" to bind on the Banggood
Original FW, while I had about 15 tries with the New FW "doing just about
the same" & failed so many times ... I though it's Game Over... So I tied
it with "DSM" where I think I have a problem with "Channels" but I saw it
was alive and continued to try on the v911

I'll try it when I'm fresh & maybe try to film it

I'm sorry - but I was not in the Hobby for 5-6years now & so much had
changed :-)

You & your other Guys did a GREAT work here - I really admire (& I had
built a few "nice things" my self in my life )

I hope I'll get better new equipment so I'll be able to enjoy it further :-)

Tal.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>