pascallanger/DIY-Multiprotocol-TX-Module #148: Note for installation of Arduino Due board

kreidler, 2018-02-12:
The new documentation is nearly perfect. Thanks.

I have had a fresh install for the STM32 and need flashing via serial.
Flashing was not possible until I added the Arduino SAM Boards to the Boards Manager.
Please add small note to Arduino_IDE_Boards.md

benlye, 2018-02-12:
The STM32 board package _should_ install the tools and dependencies it needs, not requiring the Due board package to be installed.  It has done this correctly many times for me.

What was the error you got?  Was it a fresh install of the Arduino IDE?  Which version of the IDE?  Could you try again to see if the same thing happens?

kreidler, 2018-02-12:
Windows 10 Pro / Arduino 1.8.5 / Arduino 1.0.6 was running before with some libs installed
Followed instruction given Arduino_IDE_Boards.md. Restarted machine also.

Error message "Failed to init device." like nothing connected
Tried several times with changed wires, with different serial settings until I installed Due board package.

The 4in1 boards downloaded app. 85MB on first instance. The Due board afterwards were additional app. 16MB (as far as I remember).

benlye, 2018-02-12:
'Failed to init device' is an error from stm32flash, so I take it you're trying to flash the module using an FTDI adapter?

'Failed to init device' generally does mean that the upload tool couldn't open the serial port, or couldn't detect the device on the port. I can't imagine why installing the Due board would affect that - the Due board doesn't install anything to do with flashing an STM32F1 using stmflash.exe.

Are you sure you had the board set up correctly, with the serial flashing BOOT0 jumper installed, for your initial attempts?

kreidler, 2018-02-12:
Sorry, forgot to mention that I partly followed: https://www.youtube.com/watch?v=Cygc9trueSc as the guy was flashing an older version on the same board I have and he had to install following the older instruction the Due Board.
I have had no further idea, so I installed the Due Board and only the IDE restarted it worked. BOOT0 was always jumpered. I also changed the Tx and Rx wires. 

I will test tomorrow on a second machine and try to confirm.

benlye, 2018-02-13:
That video is out of date.  The current documentation is valid for all known versions of the STM32 board.

Before we created the Arduino IDE board definition specifically for the Multi-module boards it was necessary to install several components separately.  The Due board was needed because it installed the GNU Arm Toolchain - the gcc compiler for the board.  The Multi-module STM32 board package now installs that itself.

The Due board adds nothing that would influence flashing the STM32 Multi-module via an FTDI adapter, and I can only think that you must have changed something else at the same time.

closed 2018-02-19 by pascallanger

kreidler, 2018-02-19:
Sorry was ill during last days.
Rechecked with another machine. Restart after installation of the 4in1 STM32-Board performed.
Flashing worked on frist attempt. I will try to figure out what what happened on the origin machine.

issue also closed from my side. Thanks