pascallanger/DIY-Multiprotocol-TX-Module #152: CYRF PN Code Issue

gibbi, 2018-03-22:
@pascallanger i am currently reversing DSMX to be used with an SDR for my work.. obviously the Repo was a big help on this so i'd like to share something.

```
{ 0x80, 0x69, 0x26, 0x80, 0x08, 0xF8, 0x49, 0xE7 }, 
{ 0x7D, 0x2D, 0x49, 0x54, 0xD0, 0x80, 0x40, 0xC1 },  
```

as the decoding of a genue DXE always has a single biterror when using this code so there seems to be a problem.

if i use a multitransceiver using the current code my decoder works perfectly so my code seems to replicate the cyrf just fine.

where does this table actually come from? can we maybe recheck something?

pascallanger, 2018-03-22:
Is there only a problem with these 2 lines?
These codes have been dumped from DSM TX/RX.

Pascal

gibbi, 2018-03-22:
i'am starting to think that the DXe has a messed up CYRF because every observed package has minimal biterrors. (YAY.. i take a little pride that maybe i'm the only person who would ever notice this)

 .. another genue LP5DSM is working perfectly (but it is using another column of the table so i cant rule out the tainted column theory jet). 

I'll reach out for more devices to test

gibbi, 2018-03-23:
ok, my DXe is in fact broken in a way nobody else would notice. Every symbol is supposed to be encoded in an 32 bit pattern and the very first bit of this sequence is stuck to 0 on this one. If it were an issue with the table the flips should be on different positions so no issue there.

small FYI,.. this remote is in fact using 32-8DR mode compared to 64-8DR as the code in this repo does. @pascallanger are you aware when 32-8DR is used? maybe just in the EU? My genue EU-Receiver (AR8000) works in bot 32&64 Mode depending on what i bind with.

closed 2018-03-23 by gibbi

pascallanger, 2018-03-23:
I've only seen 64-8DR so far in the dumps.