pascallanger/DIY-Multiprotocol-TX-Module #154: 4in1 module need to rebind when changing model

eupe, 2018-03-25:
Hi. Please help with an issue.
I have an irangex 4in1 module attached to a 9xrPro with ersky9x.
I have 2 planes with TGY-iA6 receivers on both plane.
I bind fine to plane A. After i bind plane B, it seems it forgets plane A. If i rebind plane A, it forgets plane B.
Therefore only the last plane bound seems to be "kept in memory".
Any idea what should be wrong?
Thank you.

pascallanger, 2018-03-25:
This is explained in Protocol Details section AFHDS2A: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#flysky-afhds2a---28
Extract of the part of interrest: RX_Num is used to give a number a given RX. **You must use a different RX_Num per RX.** A maximum of 16 AFHDS2A RXs are supported.
So basically I'm pretty sure your issue is due to the fact that you are using the same RX_Num (0 by default) for both AFHSD2A models.

closed 2018-03-25 by pascallanger

eupe, 2018-03-25:
Thank you so much. I have not hoped for such a fast answer and i posted on rcgroups as well. Sorry for that. Thanks again.