pascallanger/DIY-Multiprotocol-TX-Module #158: nRF24 transmitter power is not set to the max after binding drastically reducing range

gba-gbob, 2018-04-14:
Transmitter power is set 14dbm limiting range to about 200m.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multiprotocol.h
line 428: #define NRF_HIGH_POWER	NRF_POWER_2
 should be NRF_POWER_3 to set power to the max

pascallanger, 2018-04-21:
This is done on purpose.
By experience when the nrf is set to full power the signal appears stronger but the quality of the link is in fact highly degraded.
Test by yourself and report back with quality of signal measurements vs distance.

gba-gbob, 2018-04-21:
I did test thus the bug. 200 meters with power two and 600 meters with power 3. Started investigating after  crashing my quad twice. I am using http://www.cdebyte.com/en/product-view-news.aspx?id=109 E01-ML01DP5 which behaves as expected when power setting is applied. There are plenty bad nf24 hardware I agree and I don't mind software fix but not as default option hidden deep in the code. Add obvious configuration parameter in the config so this will not come as surprise. All the best.

pascallanger, 2018-06-01:
Done in the last commit. Will see if complains come.

closed 2018-06-01 by pascallanger