pascallanger/DIY-Multiprotocol-TX-Module #161: Hitec Red AFHSS Support

z3r0c00l2005, 2018-05-02:
Hi Guys, it woudl be great to see support for the Hitec Red protocol as used for the Ares range of indoor/park flyers (I believe this is the same as the Minima protocol)

pascallanger, 2018-05-04:
I need a Hitec transmitter and a RX to reverse the protocol...

z3r0c00l2005, 2018-05-04:
Only kit I've got is embedded in RTF Ares models.  Can hook up a logic analyser to the tx if that'll help

pascallanger, 2018-05-04:
You can always see if you have access to the spi bus between the controller and the RF component if not all integrated in one chip.
Let me know.

z3r0c00l2005, 2018-05-06:
[pwr on to bound.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/1977499/pwr.on.to.bound.zip)

I've attached a LA to the 3 non-power pins between the main controller board (Processor seems to be a ATMEGA48V-10U) and the RF module (FLDSTX1) and captured the power-on trace.

The Reciever was powered up, and the TX turned on and left to negotiate the link.

D0 is connected to FLDSTX1 pin 8
D1 to pin 6
D2 to pin 7

Other than the power pins, none of the others are soldered up so i've assumed they're unused

Attached trace is from Pulseview.

z3r0c00l2005, 2018-05-08:
[image]
[image]

Had the chance to dig further - the RF module is based upon the CC2510.

pascallanger, 2018-05-09:
The CC2510 is an all in one micro controller + RF. I'll have a look at what's going in between the main board and the rf module but little hope to find anything usefull...
We need a TX with a separated RF component. Or attack over the air which requires a lot more work...

z3r0c00l2005, 2018-05-09:
I was wandering that, especially since I didn't spot a clock signal for SPI. I was trying to assess from some of the FCC photos of there was a "budget" Hitec system that would be easier to sniff... Not sure I could see anything helpful on there. What would we need for OTA analysis?

z3r0c00l2005, 2018-05-15:
OK, I've taken the plunge and started looking Over The Air at this using a Hack RF One.

Does anyone have a GNU Radio Flow Graph for capturing FHSS systems?

looking at https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=2808483 the frequency hopping tables are known (and confirmed with a spectrum sweep) using GFSK modulation should be plausible to decode the signals.

pascallanger, 2018-05-15:
I can't access the web page link since it's asking for a username/password. 
You could try to capture only one channel for now to see if some of the serial data can be located in the frame.

z3r0c00l2005, 2018-05-15:
Odd, although the link points to the "Test Report" for FCC ID 2AEIGFLDSTX1, pages 5 & 6 seem to be the most interesting ones.  I'll try to capture a bind packet from a couple of channels later.

brandond, 2018-05-16:
FWIW, I think you have to go through the application search to get at the documents:
https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=N&application_id=UZY6Tq53n5Xm6H7eTSYdTA%3D%3D&fcc_id=2AEIGFLDSTX1

FXDUTRUCROSSET, 2018-07-07:
Pascal i live in the south of paris and can lend you a telemetry 9 ch receiver and tx module or full radio gear for a few if needed

pascallanger, 2018-07-07:
That would be cool. I work in Meudon so we should be able to meet somewhere. Please contact by PM.
FYI, Franck contacted me a while back to get the Assan protocol on multi. He didn't sent me the hardware but one of his friend did.

FXDUTRUCROSSET, 2018-07-07:
i'm on paris on friday sat sun i don't see the link for PM on git, pls contact me on modelisme.com fxdr MP

z3r0c00l2005, 2018-07-13:
I've now got access to a Spectra module, which seems to have qty 2 6-pin connectors between the main and RF boards - when i get chance i'll hook up the LA and post some traces.

FXDUTRUCROSSET, 2018-07-21:
> [image: Boxbe] <https://www.boxbe.com/overview> pascallanger (
> notifications@github.com) is not on your Guest List
> <https://www.boxbe.com/approved-list?tc_serial=41319478577&tc_rand=1032769584&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&key=iaFpZksyEza5MIZWSSpORH%2FE0pyGk3ruZWsXL3Yu3aM%3D&token=r93vKIoCgBNjAQb9BK4B8anjgVG74zrS5LhQSXkKI%2BzoHIYDvQHhTDCQGxaW1E%2Bh>
> | Approve sender
> <https://www.boxbe.com/anno?tc_serial=41319478577&tc_rand=1032769584&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&key=iaFpZksyEza5MIZWSSpORH%2FE0pyGk3ruZWsXL3Yu3aM%3D&token=r93vKIoCgBNjAQb9BK4B8anjgVG74zrS5LhQSXkKI%2BzoHIYDvQHhTDCQGxaW1E%2Bh>
> | Approve domain
> <https://www.boxbe.com/anno?tc_serial=41319478577&tc_rand=1032769584&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&dom&key=iaFpZksyEza5MIZWSSpORH%2FE0pyGk3ruZWsXL3Yu3aM%3D&token=r93vKIoCgBNjAQb9BK4B8anjgVG74zrS5LhQSXkKI%2BzoHIYDvQHhTDCQGxaW1E%2Bh>
>
> Any news?

i need to wait till u come back from hollidays in normandy?

pascallanger, 2018-07-31:
FYI, FX sent me an Hitec TX and RX (optima). I'm planning to work on this protocol in the next few days.
The TX has 2 modes to work with optima RXs or minima/micro RXs.
I don't have a minima/micro RX. If you have one available and don't mind to send it my way that would be appreciated. Otherwise if at least someone could participate in the debug seesion.

pascallanger, 2018-07-31:
I've already connected the wires on the spi bus and done my first dump. Everything looks good since I can see the cc2500 init and packets. Now the fun part starts.

z3r0c00l2005, 2018-07-31:
I've got an irange+ module and 3 models that bind with a minima protocol (although they're not genuine HiTec recievers). More than happy to help out :)

pascallanger, 2018-08-01:
Optima and Minima protocols and bind are pretty much reversed at this stage.
The telemetry for Optima is somewhat not fully there yet. I think it's because I'm missing either a sensor on the RX side so it has more data to send or on the TX side nothing is polling on the i2c bus the data. 
But telemetry is for later, I need now to put it in Multi and get what is on paper working ;-)

pascallanger, 2018-08-03:
I've got Minima "potentially" working. I was planning to push it on github today for you to test. I can share the dumps, maybe temporarely on this github for ease of access.
Optima does not work yet since I'm missing a small part of the bind negociation which I won't be able to work on until monday...

z3r0c00l2005, 2018-08-03:
Sounds good to me, I'll have the environment ready this afternoon to be able to recompile and check minima over the weekend - also well timed as i'll be able to test in a flying environment on monday as well :)

pascallanger, 2018-08-03:
I hope it will work for Minima, there is a number of unknown bytes and no RX to try out so... But since there is no negociation/talk between the TX and RX it should be ok.
I've put 2 dumps here: https://we.tl/UQFwCungJv

z3r0c00l2005, 2018-08-03:
Cool, thanks :) look forward to testing it out :)

I assume it's not going to be selectable via the OpenTX menu just yet and will use the usual AETR order on the model config

pascallanger, 2018-08-03:
On OpenTX, it will be custom protocol=39, sub-protocol=1 (for minima) and option=freq adjustment (like all frsky, corona,...)

schwabe, 2018-08-03:
@z3r0c00l2005 you can always use custom on OpenTX and use the protocol numbers.

schwabe, 2018-08-03:
@pascallanger I am adding Hitec (39) with Optima (0) and Minima (1) and cc2500 freq adj to OpenTX, right?

pascallanger, 2018-08-03:
@schwabe Yes please ;-)

pascallanger, 2018-08-03:
Ok, I can get an Optima receiver to move a servo.
Issues that I still observe:
 - if I use a random hopping table the link is not good, I'm seeing delays on the servo. This means to me that some channels are not usable, now I need to figure out which ones... The channels do not seem to be linked the ID, but there is still 3 unknown bytes which are exchanged during bind which I have no idea what to do with so still some investigations are needed.
 - telemetry packets are not received after some time. Relaunching the protocol restarts the telemetry stream but it will faill again. On the telemetry, I can get the RX voltage, TX LSSI and TX LQI. RX RSSI is not available...
Slowly but surely...

pascallanger, 2018-08-03:
@schwabe I'll do the same than AFHDS2A for the telemetry: FrSky Hub and forward to TX.
For now I'm using FrSky Hub but anyway I haven't figured out to get all the telemetry since the receiver is only sending the 1st frame on 4 (same on the original radio). It could be bacause there is no sensor attached, or the TX not having a device pulling telemetry.
So this is more a head's up that this is coming. Telemetry has GPS, temp sensors, voltage, rpm,...

pascallanger, 2018-08-03:
Ok, Optima RX is working:
 - Random ID and hopping frequency table
 - Telemetry is looks stable now, I have the RX voltage, TX RSSI and TX LQI through the FrSky Hub telemetry protocol.

TODO:
 - test on Minima -> there is a big chance that it works out of the box if I haven't done a mistake
 - Optima telemetry: how to request the next telemetry frame? Is it the RX deciding to send the frames or the TX? no idea for now...

z3r0c00l2005, 2018-08-03:
Awesome. I can take a look at the minima if you like. Is there a dev branch available?

pascallanger, 2018-08-03:
I've just made a push on the main repo.
Please test and report!

z3r0c00l2005, 2018-08-04:
Minima - 
Bound straight away

Seems to be a delay or a step on the channels, if i move the sticks to full deflection the servos respond in steps rather than a smooth motion

pascallanger, 2018-08-04:
Could you edit the hitec_cc2500.ino file and uncomment the line #define HITEC_FORCE_ID ?
It will use the ID and freq hopping of the TX I've dumped. Let me know if it improves or not.

z3r0c00l2005, 2018-08-04:
That seems to make it much worse

pascallanger, 2018-08-04:
Ok I'll see what I can do later this evening.

pascallanger, 2018-08-04:
I've been thinking about on the road... Have you tuned the frequency using option? What you describe could also be due by a bad freq tuning...

z3r0c00l2005, 2018-08-04:
Now I will admit I've not tried that since it all bound easily! What sort of values would you suggest trying?

pascallanger, 2018-08-04:
Follow the frsky/corona/sfhss freq tuning page: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md

pascallanger, 2018-08-04:
The freq tuning is module dependant so I can't give you numbers...

z3r0c00l2005, 2018-08-05:
Even tuned it doesn't appear to be smooth, although it's certainly closer to being flyable :)

pascallanger, 2018-08-05:
Have you retried once tuned  with force ID?

z3r0c00l2005, 2018-08-05:
with the forced ID ther's around a 2second delay between stick input and control deflection

z3r0c00l2005, 2018-08-05:
I can do, would I just upload a new build over serial and leave the TX connected to the PC to see that? (Usually I use a USB upload)

pascallanger, 2018-08-05:
Upload using USB, leave the module connected to USB and open the serial monitor from Arduino. You should see the module starting and giving you some information like protocol selected...

pascallanger, 2018-08-05:
You can do the same using serial just to be able to move forward... Exact same process but ftdi instead of USB.

z3r0c00l2005, 2018-08-05:
The only message dumped is for the selected protocol, nothing else in either bind or normal mode

pascallanger, 2018-08-05:
I've just published v1.2.0.25 adding the F5s in the packets.
Could you test Minima again with this version?

pascallanger, 2018-08-05:
Are you able to do a range test? Like do a sweep test on a servo and walk away with the receiver?
I really have no idea what the 2 bytes do. I've reproduced the real TX but since they vary... On my side I've used a fix value.

pascallanger, 2018-08-05:
I'll do some tests tomorrow and see if I can get a correlation with something.

pascallanger, 2018-08-05:
Yes the range option limits the module power output of any loaded protocol.
I need to check more carefully on the dumps if I can find out a correlation.

pascallanger, 2018-08-06:
Please check the latest version 1.2.0.26 which has some improvements:
 - 3 bytes TX ID
 - F5 frames same as the original TX
Still missing:
 - full telemetry

z3r0c00l2005, 2018-08-06:
Ok, on 1.2.0.25.....

Fokker binds and works well with Rx id 21....very juddery on id 01 or 11

Sopwith and spas jerky on any id, slightly better after tuning but not flyable

pascallanger, 2018-08-06:
I thought you said it was working?
What do you mean by rx id?

z3r0c00l2005, 2018-08-06:
On the bench the Fokker was ok, I'd just not tried the others yet

The Reciever number parameter in open TX

z3r0c00l2005, 2018-08-06:
I've a couple of flights in on the Fokker bound with ID 21 and it's great. I can only guess that the 3 recievers are slightly different in resilience, although they all work ok with the HiTec module. 

For testing I've been using the same model setup on the transmitter, just rebinding with the different models

pascallanger, 2018-08-06:
RX_num is between 0 and 15. That's why it sounded strange.
Have you tired 1.2.0.26 yet?

z3r0c00l2005, 2018-08-06:
I'll reflash hopefully in the next couple of days. Didn't have chance before coming flying :)

pascallanger, 2018-08-07:
Please use v1.2.027 for further testings of Minima/Micro receivers. I think I've nailed down the bug which was an overlook during implementation...

[4 more replies]

closed 2018-08-09 by pascallanger