pascallanger/DIY-Multiprotocol-TX-Module #164: Eachine E013

andremelin, 2018-05-20:
Bonjour,
J'ai depuis un an ou un peu plus la carte multiprotocol et je fais fonctionner avec cela mon eachine E010.
Que dois je faire pour le nouveau eachine E013 ? Acheter la nouvelle carte ? ou downloader la dernière version seulement ? J'ai la dernière version d'OPENTX.
Un grand merci pour votre réponse .
André Melin

pascallanger, 2018-06-01:
Le eachine E013 n'est pas supporté et ne le sera probablement jamais à cause du composant RF qu'il utilise.

Pascal

closed 2018-06-01 by pascallanger

andremelin, 2018-06-01:
Merci pour la réponse.

Pas de problème en ce qui me concerne.

Bien à vous,

 

André Melin

Dipl. Ing. Aachen

 

Avenue du Champ de la Justice, 23  

B-1325 Dion-Valmont

 

Mob.   : +3210 496 20 96 20

Tél.       : (0032)(0)10 68.95.42

 <mailto:amelin@macks.be> amelin@macks.be –  <mailto:a.melin@skynet.be> a.melin@skynet.be –  <mailto:andre@macks.be> andre@macks.be 

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus