pascallanger/DIY-Multiprotocol-TX-Module #165: Atmea328p - USBAsp upload not working

peppekerstens, 2018-05-27:
Devices/software tested: <br>
4-in-1-multiprotocol module ATMea328p based <br>
OS: Windows 10 version 1803 / build 17134.48 (physical) <br>
Programmer: [USBAsp](https://nl.aliexpress.com/item/Free-Shipping-USBasp-USB-ISP-3-3V-5V-AVR-Programmer-USB-ATMEGA8-ATMEGA128-New-10PIN-Wire/32711085670.html?spm=a2g0s.9042311.0.0.27424c4d82W2KL) <br>
Arduino version: 1.8.5 (clean install) <br>
<br>
Followed instructions: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling.md<br>
<br>
Problem: <br>
[Step 4](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling.md): avrdude throws following when uploading:<br>
<br>
Arduino: 1.8.5 (Windows 10), Board: Multi 4-in-1 (Atmega328p, 3.3V, 16MHz), No bootloader<br>
<br>
Sketch uses 24616 bytes (75%) of program storage space. Maximum is 32768 bytes.
Global variables use 811 bytes (39%) of dynamic memory, leaving 1237 bytes for local variables. Maximum is 2048 bytes.<br>
<br>
C:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -carduino -PCOM1 -b57600 -D -Uflash:w:C:\Users\peppe\AppData\Local\Temp\arduino_build_291840/Multiprotocol.ino.hex:i <br>
<br>
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53<br>
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/<br>
         Copyright (c) 2007-2014 Joerg Wunsch<br>
<br>
         System wide configuration file is "C:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"<br>
<br>
         Using Port                    : COM1<br>
         Using Programmer              : Arduino<br>
         Overriding Baud Rate          : 57600<br>
avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.<br>

peppekerstens, 2018-05-27:
**Workaround**
Based on [this](http://www.elecrom.com/avrdude-tutorial-burning-hex-files-using-usbasp-and-avrdude/):

1. Start a command prompt
2. Pick the above code-line and replace "-carduino -PCOM1 -b57600" with "-cusbasp"
3. Run...

result:
C:\Users\peppe>C:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -cusbasp -D -Uflash:w:C:\Users\peppe\AppData\Local\Temp\arduino_build_291840/Multiprotocol.ino.hex:i

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: hfuse reads as D7
avrdude: safemode: efuse reads as FD
avrdude: reading input file "C:\Users\peppe\AppData\Local\Temp\arduino_build_291840/Multiprotocol.ino.hex"
avrdude: writing flash (24616 bytes):

Writing | ################################################## | 100% 11.14s

avrdude: 24616 bytes of flash written
avrdude: verifying flash memory against C:\Users\peppe\AppData\Local\Temp\arduino_build_291840/Multiprotocol.ino.hex:
avrdude: load data flash data from input file C:\Users\peppe\AppData\Local\Temp\arduino_build_291840/Multiprotocol.ino.hex:
avrdude: input file C:\Users\peppe\AppData\Local\Temp\arduino_build_291840/Multiprotocol.ino.hex contains 24616 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 7.50s

avrdude: verifying ...
avrdude: 24616 bytes of flash verified

avrdude: safemode: hfuse reads as D7
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D7, L:FF)

avrdude done.  Thank you.

, pick the code-line above and adjust the 

When choosing the 'Multi 4-in-1 (Atmega328p, 3.3V, 16MHz)' option, the 'port' option is greyed

peppekerstens, 2018-05-27:
Probable cause
USBAsp is a 'true' USB device. Either:
- Let avrdude pick the right target by changing the programmer type to 'usbasp'
- enable the 'port' option so the right port can be chosen.

I assume this is a problem in the 'Multi 4-in-1 (Atmega328p, 3.3V, 16MHz)' board definition. I tried to reproduce the problem with a 'Arduino Nano/USBAsp' board definition combination, and this 'seems' to work (I have no actual hardware for this combination, but I could choose the 'port; to which the USBAsp programmer was attached to and the error is different (finding the programmer/port, buit failing on further communication):

Arduino: 1.8.5 (Windows 10), Board: "Arduino Nano, ATmega328P"

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xf7
Sketch uses 24984 bytes (81%) of program storage space. Maximum is 30720 bytes.
Global variables use 811 bytes (39%) of dynamic memory, leaving 1237 bytes for local variables. Maximum is 2048 bytes.
C:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\peppe\AppData\Local\Temp\arduino_build_291840/Multiprotocol.ino.hex:i 
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\peppe\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200

benlye, 2018-05-27:
The Multi-4-in-1 board definition works fine.

Step 2.4 of the instructions has you select USBasp as the programmer.  In step 4 you should choose 'Upload using programmer' rather than 'upload'.

If you choose 'upload' the Arduino IDE will incorrectly attempt to use the COM port, if you choose 'upload using programmer' it will correctly use the USBasp device.  This is just the way the IDE works.

peppekerstens, 2018-05-28:
@benlye ....confirmed... guess i should have read it a little better :|

closed 2018-05-28 by peppekerstens